arduino/Arduino #8480: Unusable WiFi101/WiFiNINA Firmware Updater

fengrun85, 2019-02-01:
I am running Windows 10 on a 4K display scaled to 250%.

The Firmware Update dialog is too small and I cannot click on the "Update Firmware" button

[image]

facchinm, 2019-02-04:
@cmaglie is making the updater DPI-aware a solution? In that case I'd move the issue to https://github.com/arduino-libraries/WiFi101-FirmwareUpdater-Plugin

fengrun85, 2019-02-12:
Any updates on this?

per1234, 2019-02-20:
Report of this issue on the Arduino Forum:
http://forum.arduino.cc/index.php?topic=599042

TimEllis2019, 2019-02-20:
@facchinm I tried your suggestion of making the updater DPI aware.  The window that comes up looks the same and the on installed in version 1.8.8 ie so small that I cannot see the button to update the firmware.  It looks like the one posted by @fengrun85 above.

facchinm, 2019-02-21:
@TimEllis2019 the DPI-awareness should be enabled at java level and is not something super easy. I'll keep all of you posted as soon as we have a solution.

TimEllis2019, 2019-02-21:
Martino,

Thank you for the personal reply.

 

I started looking at the java to make the download window.  I think I see the issue but I cannot find all of the java I need to compile.  Can you point me to

1.	package cc.arduino.plugins.wifi101;
2.	import cc.arduino.packages.BoardPort;
3.	import processing.app.Base;

 

I need the java for those packages.  If I can compile I will try and fix the issue and let you know.

-tim

facchinm, 2019-02-21:
`cc.arduino.plugins.wifi101` lives here https://github.com/arduino-libraries/WiFi101-FirmwareUpdater-Plugin .
The repo contains a build script that requires you to specify an IDE path so it retrieves the needed dependencies (`cc.arduino` and `processing.app`)  from there.

TimEllis2019, 2019-02-28:
I do not have the files on my computer.  They are supposed to be in the IDE directory.

 

REQUIRED_JARS="pde.jar arduino-core.jar jssc-2.8.0-arduino2.jar bcpg-jdk15on-152.jar bcprov-jdk15on-152.jar commons-lang3-3.3.2.jar commons-codec-1.7.jar"

 

I have version 1.8.8 of the IDE installed.  Suggestions?

facchinm, 2019-03-04:
Which operating system are you using? Jars are in either `lib` or `java` subdirectories, depending on your operating system.

TimEllis2019, 2019-03-04:
I’m using Windows 10.  I found the jars by downloading the daily build version.

 

I do note that the build script does not call out source files.

per1234, 2019-03-11:
There is now an additional report of this issue here:
https://github.com/arduino-libraries/WiFi101-FirmwareUpdater-Plugin/issues/37
I'm not sure whether this issue should be migrated to that one since it's not yet clear which repository has the code that needs to be modified.

TimEllis2019, 2019-03-11:
Without being able to update my firmware there is no way to know which issue is the cause of my problem.  Being unable to see the firmware update window is a problem.

 

arduino-libraries/WiFi101-FirmwareUpdater-Plugin#37 <https://github.com/arduino-libraries/WiFi101-FirmwareUpdater-Plugin/issues/37>

facchinm, 2019-03-11:
@fengrun85 @TimEllis2019 I'm trying to reproduce the issue with a 4k display and the scaling configuration you posted, but I'm unable to get exactly the same behaviour you get (every element is still reachable).
Anyway, I'm trying a blind patch, asking you to test it on your pc.
Pick the attached file and replace the jar file in your Arduino installation, under `./tools/WiFi101/tool/`. I'm attaching two different jars; let me know which one performs better (if it does), thanks.
[WiFi101-Updater-ArduinoIDE-Plugin-0.10.5-onlymainwindow.zip](https://github.com/arduino/Arduino/files/2952907/WiFi101-Updater-ArduinoIDE-Plugin-0.10.5-onlymainwindow.zip)
[WiFi101-Updater-ArduinoIDE-Plugin-0.10.5-allscaled.zip](https://github.com/arduino/Arduino/files/2952908/WiFi101-Updater-ArduinoIDE-Plugin-0.10.5-allscaled.zip)

TimEllis2019, 2019-03-12:
Both of the wifi101.jar files you gave me display correctly.  I downloaded firmware 1.2.1 to my board successfully using the allscalled version.  Now if I run the sketch to scan for networks it runs fine.  The AP_SimpleWebserver now correctly gets to the point it is listening for a client but I cannot talk to it via my computer and a webpage.  The webpage times out so I’m still looking into that.

 

Thank you for your help.

facchinm, 2019-03-12:
Great, thanks, I'm merging https://github.com/arduino-libraries/WiFi101-FirmwareUpdater-Plugin/commit/e3010806594501bdc405f00f5c3f6bed30d3c85f so it will be available in the upcoming IDE release

fengrun85, 2019-03-12:
> arduino-libraries/WiFi101-FirmwareUpdater-Plugin@e301080

Hi Martino,

Both files work for me.

[image: image.png]

It might be because I am using a high scale factor of 250% that you are
unable to see the same behaviour.

Thanks for fixing the problem.

Regards,
Fengrun

TimEllis2019, 2019-03-12:
Hi Fengrun,

I tried adjusting my scale factor and the display resolution but with the original file it made no difference.  Good thought though.

-tim

per1234, 2019-03-15:
@facchinm can this be closed now?

TimEllis2019, 2019-03-15:
The latest daily build worked for me.  So I’m OK to close it.

closed 2019-03-15 by facchinm