arduino/Arduino #8481: Please add an Arduino library to support the Bosch BME280 Environment sensor

SV-Zanshin, 2019-02-02:
The library is located at https://github.com/SV-Zanshin/BME280 

# BME280 library [image] [image] 

[image] *Arduino* library for using the [Bosch BME280](https://www.bosch-sensortec.com/bst/products/all_products/bme280) sensor which senses temperature, humidity and pressure. Since the package is quite small and difficult to breadboard with, the [Adafruit BME280 Breakout](https://www.adafruit.com/product/2652) board was used for development.

A complete description of the BME280 sensor can be found in the [BME280 Datasheet](https://ae-bst.resource.bosch.com/media/_tech/media/datasheets/BST-BME280_DS001-11.pdf) while the [Adafruit BME280 tutorial](https://learn.adafruit.com/adafruit-bme280-humidity-barometric-pressure-temperature-sensor-breakout) describes additional board information.

## Sensing
The BME280 allows the user to individually adjust the sampling rates for each component sensing (temperature, humidity and pressure) as well as to set the inter-measurement delay time and the [IIR](https://en.wikipedia.org/wiki/Infinite_impulse_response) filter sampling coeffient. Combining these settings allows very fine-grained control of measurement times vs. measurement accuracy.

### Temperature sensing
The BME280 measures temperatures between -40°C and 85°C with a full accuracy range between 0°C and +65°C. The accuracy is ±0.5°c with a full resolution of 0.01°C. Since the temperature sensor in the small package is tied to the PCB board on which it is soldered placed and has some self-heating as well the temperature reading will usually be above ambient.
### Humidity sensing
Non-condensing humidity between 0% and 100% is measured with an accuracy of ±3% at a maximum resolution of 0.008%.
### Pressure sensing
The pressure sensor works in temperatures between -40°C and +85°C although the zone of full accuracy only goes from 0°C to +65°C. Pressure is measured between 300hPa and 1100hPa with an accuracy of ±1.0 hPa and a resolution of 0.2Pa

A detailed library description and further details are available at the [GitHub BME280 Wiki](https://github.com/SV-Zanshin/BME280/wiki)

[image] [image]

cmaglie, 2019-02-05:
There is already a library called BME280 (we actually have a bunch of them with many variants of the name).
To fix this:
- please rename the library in `library.properties`
- commit and push the change
- remove and redo the git release tag

SV-Zanshin, 2019-02-05:
I should have thought about possible naming conflicts. I've renamed the library to "BME280_Zanshin" in the properties file and published a new release as I couldn't quite remove the old tag and re-release it.

SV-Zanshin, 2019-02-11:
Might this request have slipped through cracks? Previous responses were so prompt that I suspect this might have been the case.

stef-ladefense, 2019-02-11:
original BOSCH library : https://github.com/BoschSensortec/BME280_driver

per1234, 2019-02-11:
@SV-Zanshin cmaglie periodically goes through all the outstanding Library Manager inclusion requests and clears them all at once, then lets them build up for a little while before the next run through for the sake of efficiency. Now that the duplicate name issue has been resolved, you should be on track for the library to get added with cmaglie's next batch of requests, which shouldn't be too much longer.

SV-Zanshin, 2019-02-12:
@per1234  - no worries, I see that cmaglie comes by on a weekly basis and thought that this issue might have aged and thus escaped notice.

cmaglie, 2019-02-14:
oh yes, sorry about that, now It's added!

closed 2019-02-14 by cmaglie