arduino/Arduino #8492: arduino mini pro win 10 app: Error compiling for arduino mini pro

mfmali, 2019-02-04:
Arduino: 1.8.8 (Windows Store 1.8.19.0) (Windows 10), Board: "Arduino Pro or Pro Mini, ATmega328P (3.3V, 8 MHz)"

Build options changed, rebuilding all
C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.19.0_x86__mdqgnx93n4wtt\hardware\arduino\avr\cores\arduino\HardwareSerial.cpp: In member function 'availableForWrite':

C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.19.0_x86__mdqgnx93n4wtt\hardware\arduino\avr\cores\arduino\HardwareSerial.cpp:203:1: internal compiler error: Segmentation fault

 }

 ^

Please submit a full bug report,

with preprocessed source if appropriate.

See <http://gcc.gnu.org/bugs.html> for instructions.

lto-wrapper.exe: fatal error: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.19.0_x86__mdqgnx93n4wtt\hardware\tools\avr/bin/avr-gcc returned 1 exit status

compilation terminated.

c:/program files/windowsapps/arduinollc.arduinoide_1.8.19.0_x86__mdqgnx93n4wtt/hardware/tools/avr/bin/../lib/gcc/avr/5.4.0/../../../../avr/bin/ld.exe: error: lto-wrapper failed

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino Pro or Pro Mini.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

facchinm, 2019-02-04:
Hi @mfmali ,
thanks for the report. Are you able to reproduce the error or does it appear now and then without any apparent reason? Can you share the sketch you are trying to compile?

per1234, 2020-02-08:
Closing as fixed by the update to avr-gcc 7.3.0.

closed 2020-02-08 by per1234