arduino/Arduino #8494: Board Manager JSON (http status code from server: 403)

JorinL, 2019-02-04:
Hey there,

I receive this error in version 1.8.7 and 1.8.8:

` Error downloading http://downloads.arduino.cc/packages/package_index.json
java.lang.RuntimeException: java.lang.Exception: Fehler beim Herunterladen von http://downloads.arduino.cc/packages/package_index.json
        at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onUpdatePressed$1(ContributionManagerUI.java:150)
        at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.Exception: Fehler beim Herunterladen von http://downloads.arduino.cc/packages/package_index.json
        at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:139)
        at cc.arduino.contributions.packages.ContributionInstaller.download(ContributionInstaller.java:333)
        at cc.arduino.contributions.packages.ContributionInstaller.downloadIndexAndSignature(ContributionInstaller.java:307)
        at cc.arduino.contributions.packages.ContributionInstaller.updateIndex(ContributionInstaller.java:285)
        at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onUpdatePressed$1(ContributionManagerUI.java:146)
        ... 1 more
Caused by: java.io.IOException: Received invalid http status code from server: 403
        at cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:205)
        at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:132)
        at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:137)
        ... 5 more`

Deleted my "Arduino15" direcotry and tried also with "no proxy" selected. No change for me.

facchinm, 2019-02-04:
Hi @JorinL ,
are you able to reach  http://downloads.arduino.cc/packages/package_index.json  using a browser? If so, it looks a bit like https://github.com/arduino/Arduino/issues/7573 and you may try if PR https://github.com/arduino/Arduino/pull/8436 solves the issue.

JorinL, 2019-02-04:
> 
> 
> Hi @JorinL ,
> are you able to reach http://downloads.arduino.cc/packages/package_index.json using a browser? If so, it looks a bit like #7573 and you may try if PR #8436 solves the issue.

Hey @facchinm 
After a captcha I can reach it in my browser (firefox).
Trying now PR #8436

JorinL, 2019-02-04:
Did fix my problem - 403 persists.

facchinm, 2019-02-05:
> After a captcha I can reach it in my browser (firefox).

Wow, the captcha stuff is totally unexpected and could be causing such issues. I'm forwarding this to the web team to get some hints @mastrolinux @gvarisco

gvarisco, 2019-02-05:
@facchinm captcha is due to any of [those reasons](https://support.cloudflare.com/hc/en-us/articles/203366080-Why-do-I-see-a-captcha-or-challenge-page-Attention-Required-trying-to-visit-a-site-protected-by-Cloudflare-as-a-site-visitor-). As per [the official documentation](https://support.cloudflare.com/hc/en-us/articles/200170096-How-do-I-turn-the-Cloudflare-captcha-challenge-page-off-), I've just disabled the captcha challenge on downloads.arduino.cc/*

JorinL, 2019-02-05:
@gvarisco thank you! Works now like it should! :-)

closed 2019-02-05 by JorinL

OldSurferDude, 2020-05-17:
I'm having this problem now (2020 005 16 20h14 PDT)  Yes, I can only access the file by going through the captcha.  Can you turn it off?

BackMountainDevil, 2021-09-01:
> I'm having this problem now (2020 005 16 20h14 PDT) Yes, I can only access the file by going through the captcha. Can you turn it off? 2020 05 17 07h49 PDT Thanks! All is working again

So how you solve that?

OldSurferDude, 2021-09-01:
Sorry. I only vaguely remember the issue and I can't remember how to access
the thread.

Osd

BackMountainDevil, 2021-09-02:
Thanks for your reply. I try some methods and find out to be the Internet. I change another ISP then it work.

From An idea