arduino/Arduino #8498: [Library manager] Please update the AutoConnect library

Hieromon, 2019-02-05:
https://github.com/Hieromon/AutoConnect

I had been updated the version to v0.9.6 two month ago, but the library_index.json stills AutoConnect v0.9.5.
http://downloads.arduino.cc/libraries/library_index.json

Did I make some mistake to release a new version? If so, please tell me the wrong thing. Otherwise please update the library_index to the latest version of AutoConnect.

Thank you.

per1234, 2019-02-05:
The problem is that you forgot to update the `version` in library.properties **before** making the 0.9.6 release. At the time of the 0.9.6 release, the `version` was still set to 0.95, thus the 0.9.6 release was not picked up by the Library Manager indexer:
https://github.com/Hieromon/AutoConnect/blob/v0.9.6/library.properties#L2

closed 2019-02-05 by per1234

Hieromon, 2019-02-06:
I confirmed v0.9.6 indexed.
Thank you for your help.