arduino/Arduino #8499: Selected OTA port gets lost and then the esptool is ran against an IP

JorinL, 2019-02-05:
I experience the following problem since version 1.8.8 for windows.
I start the IDE and selecet an OTA Port, then i click on upload. It starts compiling and in the meantime the OTA device (why ever) gets los. The ide now takes esptool and changes from com1 to an ip-address.
So it really switches from espota.py to the esptool.exe

To update my devices in that situation I've build me a batchfile where I just enter the ip and the path to binary - works in 99% of the time.
So there should be no problem with my OTA devices or network.

[image]

JorinL, 2019-02-05:
Probably something to do with #5588

per1234, 2019-02-05:
It sounds like a duplicate of https://github.com/arduino/Arduino/issues/8408

Please try using the Arduino IDE Beta build:
https://www.arduino.cc/en/Main/Software#beta
and let us know whether that solves the problem.

JorinL, 2019-02-06:
Looks good so far - I only receive also the "menu duplication" like in #8408 

[image]

facchinm, 2019-02-07:
@JorinL board menu duplication is intended (one is the internal java discoverer, the other list is the pluggable one from #8038). If the list diverge, we have a bug :stuck_out_tongue: 
I'm glad 1.9.0 fixes the issue, this will greatly speedup merging the features back in master

JorinL, 2019-02-07:
Okay - I wondered about the differences "Network Port" - "Network Ports" it was no real duplicate but strange without an idea about it.

Eventually it could be named "Java Network Ports" "Pluggable Discovered Network Ports" to make it clearer for everyone else :-)

I close this one here - thanks for the fast help!

per1234, 2019-02-07:
I'll go ahead and close this as a duplicate of https://github.com/arduino/Arduino/issues/8408

We can continue any further discussion over there, if needed.

closed 2019-02-07 by per1234

JorinL, 2019-02-20:
> 
> 
> @JorinL board menu duplication is intended (one is the internal java discoverer, the other list is the pluggable one from #8038). If the list diverge, we have a bug 😛
> I'm glad 1.9.0 fixes the issue, this will greatly speedup merging the features back in master

So ehh like this diverge of the list?

[image]

facchinm, 2019-02-21:
Uuuuuh yes :slightly_smiling_face:  
The one with only one entry is the new pluggable discoverer, while the other one is the bundled java one. Which one is correct?

JorinL, 2019-02-21:
List "Network ports" is correct - the list with only one entry is much too less.