arduino/Arduino #8501: Feature request: warn about unused relational operator

i-make-robots, 2019-02-05:
**Problem**
Intend to write A=B;
Accidentally write A != B, A==B, A<=B, etc.

Currently no warning is thrown that the value returned from the operation is never used.
IMHO the nastiest off-by-one error I can make.

**Fix**
Throw a warning.
Could be suppressed for those special people who like to write A && B; instead of if(A) B;

AFAIK Java IDEs like Eclipse already warn about such behavior.  Perhaps this fix could be modeled after that one.  

Thank you!  I really appreciate all that you do for coders everywhere.

matthijskooijman, 2019-02-05:
I think you can enable this one by enabling warnings in the preferences. Would that solve your issue?

It might be good to turn these warnings on by default, but I think there's a bug report about that already.

per1234, 2019-02-05:
If you compile the following sketch with **File > Preferences > Compiler warnings** set to "More" or "All":
```c++
void setup() {
  byte A = 1;
  byte B = 2;
  A != B;
  A == B;
  A <= B;
}

void loop() {}
```
You will get the following warnings:
```
E:\\arduino\\StatementHasNoEffect\StatementHasNoEffect.ino:4:5: warning: statement has no effect [-Wunused-value]

   A != B;

     ^

E:\\arduino\\StatementHasNoEffect\StatementHasNoEffect.ino:5:5: warning: statement has no effect [-Wunused-value]

   A == B;

     ^

E:\\arduino\\StatementHasNoEffect\StatementHasNoEffect.ino:6:5: warning: statement has no effect [-Wunused-value]

   A <= B;

     ^
```

Here's the issue for having the warnings on by default: https://github.com/arduino/Arduino/issues/4184

closed 2019-02-05 by per1234

i-make-robots, 2019-02-06:
Oh, thank you!