arduino/Arduino #8502: [Library Manager] Please add munet library

domschl, 2019-02-06:
https://github.com/muwerk/munet

The network libraries (WLAN client, NTP, OTA, MQTT) on ESP boards for the cooperative scheduler muwerk from the https://github.com/muwerk project.

Thank you!

cmaglie, 2019-02-14:
Please change the name of the library because comma `,` is not allowed:

`Muwerk networking library for WLAN-Client, NTP, OTA, MQTT`

also this looks like more a description than a name, you probably want to use a more concise name like `munet` and also add a `munet.h` file to allow a better library discovery.
To fix this:

1. update library.properties and change the name field
2. commit and push
3. remove and redo the release tag.

domschl, 2019-02-15:
Library name fixed, new release 0.1.1.

domschl, 2019-02-22:
Additionally, a `munet.h` has been added to the project, release 0.1.2.

cmaglie, 2019-03-01:
Done

closed 2019-03-01 by cmaglie