arduino/Arduino #8503: [Library Manager] Please add EButton library

jonnieZG, 2019-02-06:
https://github.com/jonnieZG/EButton

Thanks.

per1234, 2019-02-07:
There are a couple of issues that need to be resolved before your libraries can be added to the Arduino Library Manager index:

Each library must have a unique `name` value in library.properties. Both https://github.com/jonnieZG/DFPlayerMini and https://github.com/jonnieZG/EWMA have the `name` value of "DFPlayerMini". Please update the library.properties of https://github.com/jonnieZG/EWMA to have a unique name.

You need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

jonnieZG, 2019-02-07:
Thanks for the detailed clarifications. I have fixed all the issues that you described, and made the releases.

jairosuse, 2019-02-08:
It’s work??

cmaglie, 2019-02-14:
Done

closed 2019-02-14 by cmaglie