arduino/Arduino #8509: Enhancement - Split Screen in the IDE

daba1208, 2019-02-07:
It would be helpful if the Arduino IDE could have a "split-screen capability, so we can see two areas of code at the same time - pretty much the same as you can do in MS Excel..

I find mysel constantly scrolling up and down to see what I have named things, for example.  Having the ability to see 2 areas of code would be great.

Nca78, 2020-02-27:
Getting highly frustrated with that too, ability to see 2 different places in the same file or two different files at the same time would save a lot of time.

BAleiHi, 2021-02-12:
Yes, a horizontal split screen would be great! (Noob here - does admin prefer that we add a new comment if it's a repeat or just like the previous one?)  Thanks.