arduino/Arduino #8511: Arduino

harshalvak, 2019-02-07:
Sketch uses 5432 bytes (2%) of program storage space. Maximum is 253952 bytes.
Global variables use 1105 bytes (13%) of dynamic memory, leaving 7087 bytes for local variables. Maximum is 8192 bytes. 

how can i solve this issue?

feikname, 2019-02-07:
_It's not an issue_, the program is just informing you the sketch size so you can have an idea of how much your code weights in relation to the selected Arduino board capacity.

In your case, 2% and 13% are well below from any need for worries :wink: 

It seems you are new to the Arduino world, next time you have a doubt like this related to your sketch, please ask in the [Arduino Forums](https://forum.arduino.cc/) first, there is the appropriate place for asking questions. This place is meant for bug reports related to the Arduino IDE only.

matthijskooijman, 2019-02-07:
Thanks for answering, @feikname. I'm gonna go ahead and close this issue.

closed 2019-02-07 by matthijskooijman