arduino/Arduino #8516: openjdk update for windows

chris3081, 2019-02-08:
Hi Guys, any chance you could release a maintenance release for Arduino on the windows store its coming up in my AV vulnerability scans https://threats.kaspersky.com/en/vulnerability/KLA11122/?utm_source=KTS_19.0.0&utm_medium=inp&utm_campaign=en

facchinm, 2019-02-11:
Hi @chris3081 ,
from the page you linked it looks like the vulnerability should only affect 
`Java SE 8 versions earlier than 8u151`.
Latest IDE (also on Windows store) ships with JRE 8u191; which version are you running?

chris3081, 2019-02-12:
Post reinstall of the IDE and a clean up across the program files in each of the drive on my machine has shown it was my machine at fault. Sorry for the false alarm it was an old install and being later in the evening time I misread the drive letter.

closed 2019-02-12 by chris3081