arduino/Arduino #8517: [Library Manager] Please add RS485HwSerial library

sauttefk, 2019-02-08:
Please add RS485HwSerial library

https://github.com/sauttefk/RS485HwSerial

sauttefk, 2019-02-12:
_ping_
Is there any obstacle that hinders the inclusion of this library?

PaulStoffregen, 2019-02-12:
Unrelated, but any chance you might consider changing the API slightly?

Two reasons:

1: Other proposals have been considered (but so far languished) by the Arduino developers for a 3rd & 4th parameter on begin(), to allow users to supply larger buffers for transmit & receive.

2: Teensy has provided this RS485 feature for years using Serial1.transmitterEnable(pin).  If you go with a separate function rather than overloading begin, and you use this same name, your library will work automatically with many RS485 projects people have made using Teensy boards.

per1234, 2019-02-12:
>Is there any obstacle that hinders the inclusion of this library?

@sauttefk everything looks fine to me. cmaglie will add it to the Library Manager index on the next run through the outstanding requests. Thanks!

sauttefk, 2019-02-12:
@PaulStoffregen Thanks for your comments!
I will change my library and go with your transmitterEnable proposal.

cmaglie, 2019-02-14:
Library added to the index.

closed 2019-02-14 by cmaglie