arduino/Arduino #8518: v1.9b error upload, must close serial monitor first

danbates2, 2019-02-09:
Seems that the serial monitor isn't closing when the upload button is pressed, causing uploads to fail as the serial monitor tried to butt in.

1.9.0-beta
I'm on 10.11.6 El Capitan.

facchinm, 2019-02-11:
Hi @danbates2 ,
I cannot reproduce on Linux. May you attach a screencast so I can try to understand where the "pause" control is failing? Thanks!

danbates2, 2019-02-11:
Hi, I don't have a screencast program. But it isn't any more complex than I describe. I must close the serial monitor to have successful uploads.
The error in the IDE (black window at bottom of sketch) is the typical upload error in red text returned when the serial port is busy and unavailable.

facchinm, 2019-02-11:
Ok, the issue was surely in beta some weeks ago (and the outcome was : press upload -> monitor was greyed out during compilation -> small reopen glitch and greyout again -> upload error), then it was patched. This is why I asked for a screenscast, but don't worry. Could you test the latest beta and report if the problem is still there?

danbates2, 2019-02-11:
Sure, where can I download or how do I compile for mac?

danbates2, 2019-02-11:
BTW, this v1.9 is a big improvement generally ;)

per1234, 2019-02-11:
@danbates2 you can download the latest beta build of the Arduino IDE here:
https://www.arduino.cc/en/Main/Software#beta
The beta build is regularly updated to reflect the ongoing development work on the IDE. It's helpful to mention the build number of the beta you're using when submitting bug reports.

For a free, open source, lightweight screencast program, I like this one:
https://www.cockos.com/licecap/

danbates2, 2019-02-12:
thanks.
my build is PR-beta1.9-BUILD-99

facchinm, 2019-02-12:
Latest is BUILD-102, so if you can redownload it and give it a try it would be awesome!

danbates2, 2019-02-12:
Still the same problem!
[image]
[image]
[image]

I close the monitor and the upload works.

facchinm, 2019-02-21:
@gvarisco could you try to reproduce this issue with any board using latest beta?

danbates2, 2019-02-21:
I've downloaded the latest again, and the issue is still present.

akash73, 2019-02-21:
Tested with 10.12.6 and latest IDE beta and correctly uploads with serial monitor open.

per1234, 2022-12-01:
Closing as "wontfix" because the Arduino IDE 1.9.x effort has been abandoned so that the resources can be put to Arduino IDE 2.x instead.

closed 2022-12-01 by per1234