arduino/Arduino #8521: https://github.com/rleddy/tinycmdtable/tree/v1.0

rleddy, 2019-02-10:
Just a little library for a little table. Plus some short node.js tools for generating most of the sketch that uses the commands.

per1234, 2019-02-11:
In order to be added to the Arduino Library Manager index, your library must define the `url` field in library.properties. In your repository, that field is currently blank. Please add a URL and then either create a new tag/release or else redo the existing one.

Once that's finished, please comment here so we'll know that we can proceed.

Reference:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

rleddy, 2019-02-12:
The url field has been added

per1234, 2019-02-12:
You still need to create a release or tag with the fixed library.properties.

rleddy, 2019-02-12:
Done

cmaglie, 2019-02-14:
Added the library to the index.

closed 2019-02-14 by cmaglie