arduino/Arduino #8525: WiFi101.h WiFi.disconnect() function seems broken

rjbuckley2, 2019-02-11:
void loop() {
  while ( status != WL_CONNECTED ) 
  {
    Serial.println("Obviously not connected...");
    status = WiFi.begin(ssid);
  }
  Serial.println("Connected. Trying to disconnect");
   WiFi.disconnect();
   delay(30000);   
}

This code prints:
11:14:19.204 -> Obviously not connected...
11:14:21.742 -> Connected. Trying to disconnect
11:14:51.757 -> Connected. Trying to disconnect
11:15:21.727 -> Connected. Trying to disconnect
 Trying to disconnect ad infinitum

Rocketct, 2019-02-15:
hi @rjbuckley2 i have tested your code it seem that work as expected the ```status``` valriable is never update after the while loop and inside the ```WiFi.disconnect()`` too, then after the first cycle it never goes inside the connection while.
The loop function is like a while(1), than loop forever due to this it follow to call the serial print and the disconnect.

rjbuckley2, 2019-02-15:
Thank you for your help!
-rb

Rocketct, 2019-02-15:
welcome :)

closed 2019-02-15 by Rocketct

rjbuckley2, 2020-05-25:
---------- Forwarded message ---------
From: Casey Brinck <cbrinck@secular.org>
Date: Mon, May 25, 2020 at 9:00 AM
Subject: A veteran's take on Memorial Day
To: <rjbuckley.music@gmail.com>

I want to share some of the “truths” I discovered during my time spent
serving in our nation’s military.
<https://secular.us6.list-manage.com/track/click?u=9241bef44976519ab9f9dff94&id=280b0d37e5&e=8e6b4b8955>

As we mark Memorial Day this year, and remember those who have given the
ultimate sacrifice to uphold the ideals and freedoms we all hold so dear,
let us remember why they fought and died.

During my time in the U.S. Army, I learned how deeply my fellow soldiers
believe in our nation and its values. However, I also discovered a number
of hard “truths”—starting with the oath we swear when enlisting—that I want
to share with you today.
*I, (NAME), do solemnly swear (or affirm) that I will support and defend
the Constitution of the United States against all enemies, foreign and
domestic; that I will bear true faith and allegiance to the same; and that
I will obey the orders of the President of the United States and the orders
of the officers appointed over me, according to regulations and the Uniform
Code of Military Justice. So help me God.*

Did you catch it? Kind of hard to miss right? Despite the religious
declaration at the end, when I said those words, I meant them
wholeheartedly. It seemed to me that everyone else around me meant them
too...the exuberance and naivety of youth I suppose.

While serving, I had the honor of participating in Operation Restore Hope
(in Mogadishu, Somalia). I experienced many different things during my time
in Africa that have stuck with me to this day—nearly 30 years later. One
occurred on the second day I was in-country. My tank crew and I were
pulling guard duty at one of the checkpoints that ran between the city and
our makeshift base on the western outskirts of Mogadishu.

I was sitting on the front slope of our M1A1 MBT when a small boy around
seven, the same age my son is now, approached me and said, “Hey America
(what they called U.S. soldiers), what is your name?”

I replied, “Casey, what is your name?” He said, “My name is Muhammad, and
you should have a proper Muslim name.” After some careful thought, he
looked up at me and said, “Your name is now Muhammad, too.”
*Casey, sitting on an M1A1 in 1992.*

I laughed and said, “thank you,” as he started into his pitch on all of the
things he could sell to us, from local crafts to marijuana.

I got to know Muhammad over the next several months. He would come by and
speak to my tank crew and I (always referring to me as Muhammad) whenever
we were guarding that particular checkpoint. I learned a lot from him,
including that despite our differences, we are all very much the same, and
that true friendships can be formed between even the least likely of
individuals.

Now, I recognize that this isn't any great revelation in the context of the
secular (freethought) movement, but for me, *during my time in Somalia, I
also learned that everyone’s God is the “right” God. *No matter who you
speak to, those who are devout in their faith are always convinced it is
they who are correct.

Whether it was the soldier from Mississippi who carried his Bible with him
everywhere he went or the RPG wielding Somali who carried his prayer rug
with similar frequency and conviction, they were all certain that God
existed and that they knew who he (it’s usually a male god) is.

Experiencing this contradiction firsthand on a daily basis sparked one of
my earliest self-recognitions as an atheist. It was clear to me that they
were both so blind to any alternative perspective that neither could be
[... 33 lines cut ...]

Casey Brinck
*Director of Policy & Government Affairs*
*Secular Coalition for America*
[image: Facebook]
<https://secular.us6.list-manage.com/track/click?u=9241bef44976519ab9f9dff94&id=34e0174d55&e=8e6b4b8955>
[image: Twitter]
<https://secular.us6.list-manage.com/track/click?u=9241bef44976519ab9f9dff94&id=e89a972c8f&e=8e6b4b8955>
[image: Link]
<https://secular.us6.list-manage.com/track/click?u=9241bef44976519ab9f9dff94&id=43ae4014bb&e=8e6b4b8955>
[image: Website]
<https://secular.us6.list-manage.com/track/click?u=9241bef44976519ab9f9dff94&id=a205ab87b3&e=8e6b4b8955>
*Copyright © 2020 Secular Coalition for America, All rights reserved.*
*1012 14th St, NW, Suite 205, Washington, DC 20005 *

Contributions to the Secular Coalition for America are not tax-deductible;
they support our effective, citizen-based advocacy and lobbying efforts.
Tax-deductible donations, including from Donor Advised Funds and IRA
Rollovers, can be made to the Secular Coalition for America Education Fund.
You are receiving this email because you requested to receive updates from
the Secular Coalition for America.
Manage subscription
<https://secular.us6.list-manage.com/profile?u=9241bef44976519ab9f9dff94&id=4e255484ff&e=8e6b4b8955>
 | Unsubscribe
<https://secular.us6.list-manage.com/unsubscribe?u=9241bef44976519ab9f9dff94&id=4e255484ff&e=8e6b4b8955&c=ba3a2aab99>