arduino/Arduino #8528: Bug. Exception Type: EXC_BAD_ACCESS (SIGABRT)

danbates2, 2019-02-13:
Crashlog on Mac to follow. Enjoy!
```
Process:               Arduino [39285]
Path:                  /Applications/Arduino.app/Contents/MacOS/Arduino
Identifier:            cc.arduino.Arduino
Version:               PR-beta1.9-BUILD-102 (1.8.9)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           Arduino [39285]
User ID:               501

Date/Time:             2019-02-13 10:54:24.908 +0000
OS Version:            Mac OS X 10.11.6 (15G22010)
Report Version:        11
Anonymous UUID:        AD39786B-25A2-8395-D957-0F1DA5460E84

Sleep/Wake UUID:       2066D595-1FFA-4BBC-94B5-6B8637125C91

Time Awake Since Boot: 100000 seconds
Time Since Wake:       4700 seconds

System Integrity Protection: enabled

Crashed Thread:        24  Java: Java2D Queue Flusher

Exception Type:        EXC_BAD_ACCESS (SIGABRT)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000002
Exception Note:        EXC_CORPSE_NOTIFY

VM Regions Near 0x2:
--> 
    __TEXT                 00000001007ad000-00000001007b4000 [   28K] r-x/rwx SM=COW  /Applications/Arduino.app/Contents/MacOS/Arduino

Application Specific Information:
abort() called

Thread 0:: AppKit Thread  Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff9d24ef72 mach_msg_trap + 10
1   libsystem_kernel.dylib        	0x00007fff9d24e3b3 mach_msg + 55
2   com.apple.CoreFoundation      	0x00007fff94616114 __CFRunLoopServiceMachPort + 212
3   com.apple.CoreFoundation      	0x00007fff946155dc __CFRunLoopRun + 1356
4   com.apple.CoreFoundation      	0x00007fff94614e28 CFRunLoopRunSpecific + 296
5   com.apple.HIToolbox           	0x00007fff8ba34935 RunCurrentEventLoopInMode + 235
6   com.apple.HIToolbox           	0x00007fff8ba3476f ReceiveNextEventCommon + 432
7   com.apple.HIToolbox           	0x00007fff8ba345af _BlockUntilNextEventMatchingListInModeWithFilter + 71
8   com.apple.AppKit              	0x00007fff9656adf6 _DPSNextEvent + 1067
9   com.apple.AppKit              	0x00007fff9656a226 -[NSApplication _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 454
10  libosxapp.dylib               	0x0000000117360328 -[NSApplicationAWT nextEventMatchingMask:untilDate:inMode:dequeue:] + 124
11  com.apple.AppKit              	0x00007fff9655ed80 -[NSApplication run] + 682
12  libosxapp.dylib               	0x000000011735ff72 +[NSApplicationAWT runAWTLoopWithApp:] + 156
13  libsplashscreen.dylib         	0x000000011733c5c8 __SplashInitPlatform_block_invoke_1 + 360
14  JavaNativeFoundation          	0x00000001173805f5 +[JNFRunLoop _performCopiedBlock:] + 20
15  com.apple.Foundation          	0x00007fff9b1b2dbe __NSThreadPerformPerform + 279
16  com.apple.CoreFoundation      	0x00007fff946367e1 __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 17
17  com.apple.CoreFoundation      	0x00007fff94615f0c __CFRunLoopDoSources0 + 556
18  com.apple.CoreFoundation      	0x00007fff9461542f __CFRunLoopRun + 927
19  com.apple.CoreFoundation      	0x00007fff94614e28 CFRunLoopRunSpecific + 296
20  libjli.dylib                  	0x0000000100a1d8fe CreateExecutionEnvironment + 871
21  libjli.dylib                  	0x0000000100a194cc JLI_Launch + 1952
22  cc.arduino.Arduino            	0x00000001007b1aa3 launch + 14627
23  cc.arduino.Arduino            	0x00000001007adf5f main + 175
24  cc.arduino.Arduino            	0x00000001007adea4 start + 52

Thread 1:: Dispatch queue: com.apple.libdispatch-manager
0   libsystem_kernel.dylib        	0x00007fff9d255efa kevent_qos + 10
1   libdispatch.dylib             	0x00007fff91414165 _dispatch_mgr_invoke + 216
2   libdispatch.dylib             	0x00007fff91413dcd _dispatch_mgr_thread + 52

Thread 2:
0   libsystem_kernel.dylib        	0x00007fff9d25510a __semwait_signal + 10
[... 1413 lines cut ...]
TOTAL, minus reserved VM space     4.2G     2264 

Model: MacBookPro8,2, BootROM MBP81.0050.B00, 4 processors, Intel Core i7, 2.2 GHz, 8 GB, SMC 1.69f3
Graphics: Intel HD Graphics 3000, Intel HD Graphics 3000, Built-In
Graphics: AMD Radeon HD 6750M, AMD Radeon HD 6750M, PCIe, 512 MB
Memory Module: BANK 0/DIMM0, 4 GB, DDR3, 1333 MHz, 0x859B, 0x435435313236344243313333392E4D313646
Memory Module: BANK 1/DIMM0, 4 GB, DDR3, 1333 MHz, 0x859B, 0x435435313236344243313333392E4D313646
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0xD6), Broadcom BCM43xx 1.0 (5.106.98.102.30)
Bluetooth: Version 4.4.6f4, 3 services, 19 devices, 1 incoming serial ports
Network Service: Ethernet, Ethernet, en0
Network Service: Wi-Fi, AirPort, en1
Serial ATA Device: SanDisk SD8TB8U256G1001, 256.06 GB
Serial ATA Device: HL-DT-ST DVDRW  GS31N
USB Device: USB 2.0 Bus
USB Device: Hub
USB Device: IR Receiver
USB Device: USB 2.0 Bus
USB Device: FaceTime HD Camera (Built-in)
USB Device: Hub
USB Device: Apple Internal Keyboard / Trackpad
USB Device: BRCM2070 Hub
USB Device: Bluetooth USB Host Controller
Thunderbolt Bus: MacBook Pro, Apple Inc., 25.1

```

matthijskooijman, 2019-02-13:
Can you tell us a bit about what you were doing when this happened? Also, is it reproducible, or have you at least seen it more than once?

The backtrace looks like java internals, so I wouldn't be surprised this is a java, system or driver bug, rather than something we can fix in Arduino..

danbates2, 2019-02-13:
Could be just java.
I was doing nothing with arduino at the time and all it's windows were
closed.

danbates2, 2019-02-17:
i'll reopen if it happens again ever.

closed 2019-02-17 by danbates2