arduino/Arduino #8529: [Feature Req] Serial Monitor with 'Send RST' button.

danbates2, 2019-02-13:
Just in the same way a reset is issued at the end of an upload, it'd be neat to have a reset command available in the serial monitor I think.

danbates2, 2019-02-13:
The reset used to occur when selecting the baud rate, however this seems to of changed in beta:
PR-beta1.9-BUILD-102

I think it's a good idea to not reset on baud rate selection. Hence a button replacing that feature would be shaweet.

per1234, 2019-02-13:
Closing as duplicate of https://github.com/arduino/Arduino/issues/4353

We welcome bug reports and feature requests but in the future please take the time to search for existing open and closed issues and pull requests before submitting an issue:
https://github.com/arduino/Arduino/issues?&q=
If you do find the issue has already been reported or a fix has been submitted, you are welcome to give it a "thumbs up" by clicking on the smiley face icon at the top right to show your support. Please only comment on the issue if you have additional information to add.

closed 2019-02-13 by per1234