arduino/Arduino #8533: Arduino 1.8.8 crashes when trying to open it on MacOS Mojave 10.14.3.

HunterRB, 2019-02-13:
Arduino 1.8.8 downloaded no problem and worked the first few times then randomly started crashing when I try to open it. It will load all the way till it says "starting" then just stops. I've tried deleting it and redownloading it and also tried multiple older versions and nothing has work. Anyones help would be greatly appreciated. Thank you.

facchinm, 2019-02-14:
Hi @HunterRB ,
can you try launching it from a Terminal as `Arduino.app/Contents/MacOS/Arduino` and paste here the console logs? Thanks

gvarisco, 2019-02-14:
Should be _`/Applications/Arduino.app/Contents/MacOS/Arduino`_ in case you have troubles finding the folder.

HunterRB, 2019-02-14:
Pretty new to all this and am wondering what you mean by a Terminal and console logs

HunterRB, 2019-02-14:
Figured out and this error appeared. open -a /applications/Arduino.app/Contents/MacOS/Arduino
FSPathMakeRef(/applications/Arduino.app/Contents/MacOS/Arduino) failed with error -43.

HunterRB, 2019-02-18:
Tried it again and it says "no such file or directory"

HunterRB, 2019-02-19:
Tried redownloading it randomly and works fine they must've fixed the issue

closed 2019-02-19 by HunterRB