arduino/Arduino #8534: Problem to read string from serial

Gionni67, 2019-02-14:
I have write the code below in arduino ide 1.8.8. When I execute the code and type a string in the serial monitor, all the chars has printedexcluded 'p' char. 
'p' char is always lost. 
What do you thinks?
Thanks for replyl.

#include <EEPROM.h>
struct st {
  char a[17];
  char b[26];
  char c[26];
  byte d;
  float e;
  float f;
  float g;
  float h;
  byte i;
};

void setup() {
  Serial.begin(9600);
  while (!Serial) {};
}

void loop() 
{
  if(Serial.available())
  {
    String cmd=Serial.readStringUntil("\n");
    Serial.println(cmd);
    if(cmd.equals("write"))
    {
      st s1 = { "", "", "", 1, 1.0f, 1.0f, 1.0f, 1.0f, 1 };
      EEPROM.put(0, s1);
    }
  }
}

per1234, 2019-02-14:
The problem is that you are passing `Serial.readStringUntil()` the string `"\n"` (which happens to have the value of 'p'), rather than the character `'\n'`.

If you turn warnings on at **File > Preferences > Compiler Warnings**, you will get a useful warning about this:
```
C:\Users\per\AppData\Local\Temp\arduino_modified_sketch_50020\sketch_feb13a.ino:23:45: warning: invalid conversion from 'const char*' to 'char' [-fpermissive]

     String cmd = Serial.readStringUntil("\n");

                                             ^
```

closed 2019-02-14 by per1234

Gionni67, 2019-02-15:
With your correction all character as prindet corretly.
Thank you very much.