arduino/Arduino #8535: Save sketch recreate the sketch file and break the hard link

45gfg9, 2019-02-14:
OS: macOS 10.14.3
IDE: 1.8.8
We create a hard link to a sketch: 
``` sh
ln ~/Documents/Arduino/test/test.ino ~
ls -li ~/Documents/Arduino/test/test.ino ~/test.ino
```
```
12898678091 -rw-r--r--  2 45gfg9  staff  225  2 14 13:21 /Users/45gfg9/Documents/Arduino/test/test.ino
12898678091 -rw-r--r--  2 45gfg9  staff  225  2 14 13:21 /Users/45gfg9/test.ino
```
Go to IDE and change the sketch, then
`ls -li ~/Documents/Arduino/test/test.ino ~/test.ino`
```
12898678261 -rw-r--r--  1 45gfg9  staff  221  2 14 13:23 /Users/45gfg9/Documents/Arduino/test/test.ino
12898678091 -rw-r--r--  1 45gfg9  staff  225  2 14 13:21 /Users/45gfg9/test.ino
```

matthijskooijman, 2019-04-02:
Are you sure this issue is fixed? I can see that the code would fix *symlinks*, but this issue is about hardlinks (which is tricky to fix with the current write-then-rename approach, I think to work with hardlinks you'll have to write directly to the original file, with the risk of half-overwriting it on errors such as disk full).

45gfg9, 2019-04-03:
Same... I can't sure whether it's fixED or not cuz both the build && the branch was deleted. You said it's about symlinks but not hard links :( I'll build one myself later.

per1234, 2019-04-03:
@45gfg9 the test builds were deleted for https://github.com/arduino/Arduino/pull/8744 because https://github.com/arduino/Arduino/pull/8744 has been merged into the `master` branch, and thus is now available for testing in the hourly build of the Arduino IDE:
https://www.arduino.cc/en/Main/Software#hourly

cmaglie, 2019-04-03:
> I can see that the code would fix symlinks, but this issue is about hardlinks 

I see, I don't think it's fixed.

Humancell, 2019-06-18:
I'm curious what Release or version this might be fixed in?  I'd like to test it.  I've tried using symlinks and hardlinks on Windows (using mklink) and in both cases when I save a sketch file it is deleting the link and then creating a new file in it's place.

I'm guessing the fix being worked on would follow the link and save over the original file?

BTW ... this would be a REALLY great feature I've been looking for!

MartyMacGyver, 2019-07-16:
Same issue on Windows 10 - soft and hard symlinks work great right up until you save - then Arduino decides to completely replace the link with a new file. There's no useful reason for this behavior - Arduino can plainly read from a symlink, and writing should be no stretch - and it makes library development unnecessarily difficult.

MartyMacGyver, 2019-07-17:
This is the part that makes zero sense to me... normally if you're saving a file and it exists, you open it, write to it, close it, and you're done.

Instead, the IDE does a strange dance of _deleting the file first, then replacing it with what was a temp file._

https://github.com/arduino/Arduino/blob/fb9ce0ade323f6f36793f29d615db1272d184ba4/arduino-core/src/processing/app/BaseNoGui.java#L903

How about, instead of locating the original file and destroying it and therefore breaking its symlinks, the IDE simply updates the file in place in a normal way?

MartyMacGyver, 2019-07-17:
And this appears to be the second attempt to fix the same problem:

https://github.com/arduino/Arduino/issues/5478

The original problem appears to have started about 10 years ago as part of a sync from Processing.

https://github.com/arduino/Arduino/commit/2e26a2d994d6a082f4a83ddc7bcdb7f29d9b69b6#diff-b6fda0bf6be0a53c0741a9b02ac16f37R1902

The problem persists in Processing to this day:

https://github.com/processing/processing/blob/b9049243bd74c8a6395f102944f04ca5196ec43d/app/src/processing/app/Util.java#L193

Is there any will to actually fix the underlying problem in a way that necessarily fixes Processing's broken precedent in this regard?

lmihalkovic, 2019-07-18:
Yup.. familiar song ... one of the many idiosyncrasies I had fixed in my branch.. the whole ‘file system’ layer is there but so well hidden that the devs used to be convinced it was too much work to crate one, and consequently accept the status quo.

sirtet, 2019-09-05:
Just ran into this problem now with 1.8.9 on Win10_64

PR #8744 (2019-04-02): Follow symlinks when saving sketch
Fixes #8535
commit 05de5c95 (2019-04-02): Follow symlinks when saving sketch

Fixes #8535