arduino/Arduino #8538: Wire Library not recognized

XM4r1u5, 2019-02-15:
So when I try to include the Wire.h library the text 'Wire' isn't fat like eg. SD or SPI as in the picture,
I reinstalled the ide, tried on my laptop and downloaded older versions of the ide, my whole code worked until I updated my PC.
thx for any help
[image]

per1234, 2019-02-15:
You shouldn't put much importance on the way things are colored in the Arduino IDE. This is only a matter of if and how the library author decided to define the keywords for highlighting in the Arduino IDE. That is purely an aesthetic matter and has no functional effect on the code.

Especially in 3rd party libraries, you'll find that many authors don't even bother to define keywords and others made an attempt but did it wrong or incompletely. The Arduino IDE also highlights every keyword for every installed library, so once you get a lot of libraries installed you'll find that practically every other word ends up highlighted, even if most of those keywords are from libraries you're not even using in that sketch.

Here's the cause of the different highlighting between the word "SPI" and the word "Wire" (the same applies for "SD"). Here's the SPI library's definition of the keyword "SPI":
https://github.com/arduino/ArduinoCore-avr/blob/1.6.23/libraries/SPI/keywords.txt#L9
```
SPI	KEYWORD1
```
Here's the Arduino AVR Boards Wire library's definition of the keyword "Wire":
https://github.com/arduino/ArduinoCore-avr/blob/1.6.23/libraries/Wire/keywords.txt#L25
```
Wire	KEYWORD2
```
You can see that the author of the SPI library chose the KEYWORD_TOKENTYPE of `KEYWORD1` for the "SPI" keyword while the author of the Wire library chose the KEYWORD_TOKENTYPE of `KEYWORD2` for the "Wire" keyword.

I do think `KEYWORD1` would be a more appropriate choice, but the keywords system has historically been not well documented so people made their own interpretations of what the meanings of the various KEYWORD_TOKENTYPEs are. The official Arduino libraries at least should be consistent in their keyword definitions so I would consider this a minor bug that ought to be fixed (will need to be done in every hardware core).

More information on the Arduino IDE's keywords system:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#keywords

sandeepmistry, 2019-09-16:
@per1234 should this be transferred to the `ArduinoCore-avr` repo?

per1234, 2019-09-16:
@sandeepmistry if so, then duplicate issues should also be opened in `ArduinoCore-sam`, `ArduinoCore-samd`, and `ArduinoCore-megaavr` repositories as well, since the same issue occurs in their Wire libraries.

The issue does not occur in `ArduinoCore-arc32` or `ArduinoCore-nRF528x-mbedos` repositories because those Wire libraries don't even have a keywords.txt file, which could be considered its own bug.

If you agree with my idea to change the keyword type from `KEYWORD2` to `KEYWORD1`, I'd be happy to just submit PRs to each of the repos so we can get this closed.

sandeepmistry, 2019-09-16:
@per1234 pull requests would be great!

per1234, 2022-04-19:
Closing as resolved as the keyword definition has now been updated in all the relevant official boards platforms.

- https://github.com/arduino/ArduinoCore-avr/pull/404
- https://github.com/arduino/ArduinoCore-megaavr/pull/99
- https://github.com/arduino/ArduinoCore-samd/pull/619

closed 2022-04-19 by per1234