arduino/Arduino #8541: [Library manager] Please add sqlite3

tobozo, 2019-02-15:
Hello,

can you please add this project to the library manager:

https://github.com/siara-cc/esp32_arduino_sqlite3_lib/releases

thanks ^_^

cmaglie, 2019-03-01:
Done

closed 2019-03-01 by cmaglie

siara-cc, 2019-03-01:
@cmaglie Hi, Thank you for adding our library.  It seems it has taken the very first release for Zip download.. Is it possible to add the latest release? https://github.com/siara-cc/esp32_arduino_sqlite3_lib/releases/tag/v1.2

[image]

tobozo, 2019-03-01:
@siara-cc the Arduino Library Manager only trusts the version in your own [library.properties](https://github.com/siara-cc/esp32_arduino_sqlite3_lib/blob/master/library.properties#L2) file

tobozo, 2019-03-01:
you'll need to create a new tag+release with the updated properties file, the Library Manager will detect it and send an update signal on next propagation (few hours)

siara-cc, 2019-03-01:
@tobozo Thank you so much..  I have done that.

per1234, 2019-03-01:
@siara-cc the `version` value of `v1.3` in your library.properties file [at your v1.3 release](https://github.com/siara-cc/esp32_arduino_sqlite3_lib/blob/v1.3/library.properties#L2) is not compliant with the library.properties specification:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format

This will cause your v1.3 release to not be picked up by the Library Manager indexer job.

You can name your tags/releases however you like, but the `version` value in library.properties at the time of the tag must be compliant with the specification. It is the library.properties `version` value that will be used by Library Manager. The specification is a "relaxed semver" to allow compatibility with a wider range of versioning preferences, so this does permit the `1.3` style of versioning you appear to prefer. If you want a well defined versioning specification that is library.properties compliant, you can follow semver:
http://semver.org/

siara-cc, 2019-03-01:
@per1234 Thank you for alerting me.  I have now corrected this problem.