arduino/Arduino #8543: Incorrect compilation errors with TABs since Arduino IDE release 1.8.6

EJKrupe, 2019-02-15:
Unfortunately I can't go beyond version 1.8.5 of the Arduino IDE because all later versions may cause incorrect compilation errors when using TABs. When I merge the code that I have divided in 6 tabs in one file it also works for the later IDE versions. Starting with IDE 1.8.6. I for example may get errors regarding functions that 'were not declared in this scope'. Different kind of compilation errors occur (depending on the use of IDE 1.8.6 or 1.8.8) when I compile Marlin releases 1.1.8 or 1.1.9 for my 3D Printer. Fortunately also here older IDE releases like 1.8.2, 1.8.4 and 1.8.5 work fine. I have this problem on two different systems (HP laptop and Microsoft Surface) running on Windows 10.

per1234, 2019-02-16:
Please test using the hourly build of the Arduino IDE and then let us know if the issue still occurs:
https://www.arduino.cc/en/Main/Software#hourly

sergikboy, 2019-02-20:
Up! I am use version 1.8.9 hourly build 2019/02/04 10:33 and windows 10 64 bit. same problem. Only 1.8.5. and previous versions working is correctly.

per1234, 2019-02-20:
@sergikboy thanks for testing with the latest hourly build!

Please provide a specific set of instructions that we can use to reproduce the issue. Ideally this would be a [Minimal, Complete, and Verifiable example](https://stackoverflow.com/help/mcve).

sergikboy, 2019-02-21:
`In function 'global constructors keyed to 65535_0_MegaCentral.ino.cpp.o':

lto1.exe: internal compiler error: Segmentation fault

Please submit a full bug report,

with preprocessed source if appropriate.

See <http://gcc.gnu.org/bugs.html> for instructions.

lto-wrapper.exe: fatal error: C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avr-gcc returned 1 exit status

compilation terminated.

c:/program files (x86)/arduino/hardware/tools/avr/bin/../lib/gcc/avr/5.4.0/../../../../avr/bin/ld.exe: error: lto-wrapper failed

collect2.exe: error: ld returned 1 exit status`

sorry but its impossible. I do not know whether this is due to Tabs. My project consists of several files. I just press CTRL + R and play the lottery. compilation result: error or successful. I do not change anything in the sketch. just press CTRL + R. There is no dependency. compilation results are different:
successful, error, successful, successful, error, error, error.
I can provide my project for research.

per1234, 2019-02-21:
@sergikboy that issue is covered in this issue report: https://github.com/arduino/Arduino/issues/7949

You're right that it's impossible to provide instructions to reproduce the issue. I've spent a lot of time trying but the error comes and goes at random. It's not directly related to tabs. It also doesn't really have anything to do with the IDE version, other in that each IDE version comes with a specific version of Arduino AVR Boards. However, Boards Manager (Tools > Board > Boards Manager) allows us to install any Arduino AVR Boards version with any IDE version, so that is not really relevant.

#### Here's the traditional workaround:
1. Tools > Board > Boards Manager
1. Wait for downloads to finish.
1. When you move the mouse pointer over "Arduino AVR Boards", you will see a "Select version" dropdown menu appear. Select "1.6.21".
1. Click "Install".
1. Wait for installation to finish.
1. Click "Close".

Due to a [bug](url=https://github.com/arduino/Arduino/issues/7973), this workaround doesn't work with Arduino IDE 1.8.6, but it will work with any other version of the Arduino IDE.

If you have **File > Preferences > Check for updates on startup** checked, the Arduino IDE may occasionally notify you that a new version of Arduino AVR Boards is available, you'll need to refrain from updating back to the new Arduino AVR Boards version, otherwise you'll be back to seeing the segmentation fault error again.

---
#### Alternate workaround:
One of the Arduino developers [recently worked on this and has provided a potential fix by updating avr-gcc to 7.3.0](https://github.com/arduino/Arduino/issues/7949#issuecomment-442969335). There is a special beta testing hardware package you can install via the Arduino IDE's Boards Manager that has the fix:
1. (In the Arduino IDE) File > Preferences
1. In the "Additional Boards Manager URLs" field, enter: http://downloads.arduino.cc/packages/package_avr_7.3.0_index.json If you already have an additional Boards Manager URL in that field, you can separate multiple URLs with commas or click the button on the right side of the field to open a dialog that allows you to add each on its own line.
1. Click "OK"
1. Tools > Board > Boards Manager
1. Wait for the downloads to finish
1. Click on "Arduino AVR Boards by Arduino"
1. Click the "Update" button.
1. Wait for the update to finish.
1. Click "Close".

---
@EJKrupe we're still hoping to hear back from you on whether your issue still occurs when you test with the latest hourly build of the Arduino IDE.

MartyMacGyver, 2019-09-08:
That certainly fixes the problem - when will this fix be part of the official toolchain here?

per1234, 2019-09-16:
Closing due to lack of feedback.

closed 2019-09-16 by per1234

MartyMacGyver, 2019-09-16:
So does that mean you're now using this or a later compiler in the nightly builds?

per1234, 2019-09-16:
@MartyMacGyver yes. avr-gcc 7.3.0-atmel3.6.1-arduino5 is used in the nightly builds and even in Arduino IDE 1.8.10 in the bundled Arduino AVR Boards 1.8.1. The only thing preventing https://github.com/arduino/Arduino/issues/7949 from being considered completely fixed is that there has been no release of Arduino AVR Boards 1.8.1 to the Boards Manager index.

twanek, 2019-12-12:
hello!

i already use ide 1.8.10, but the issue is still there :(  with avr boards manager 1.8.2
what should i try?