arduino/Arduino #8545: SPI.begin() does not set Clock HIGH in SPI_MODE3

sebasqure, 2019-02-15:
If SPI settings are set to SPI_MODE3 clock signal should idle at HIGH and before doing SPI.beginTransaction(LTC6811_SPI_config); being LTC6811_SPI_config = SPISettings LTC6811_SPI_config(SPISettings(500000, MSBFIRST, SPI_MODE3)); clock is LOW. 

The problem of this  behaviour is that the first message is ignored. After doing SPI.beginTransaction(LTC6811_SPI_config); for the first time the clock will start idling at HIGH as expected.

sebasqure, 2019-02-17:
Actually I now think that this is the intended behaviour because SPI is a multi device bus protocol. 
See [https://forum.pjrc.com/threads/55183-SPI-begin()-does-not-set-Clock-HIGH-in-SPI_MODE3?p=198395#post198395](url)

sandeepmistry, 2019-09-16:
Hi @sebasqure,

What board are you seeing the issue on?

Also, based on your last comment can this issue be closed?

sebasqure, 2019-09-20:
Hi @sandeepmistry ,

The board is a Teensy 3.2.

If you think that this is the intend behaviour, yes you can close it.

sandeepmistry, 2019-09-23:
Hi @sebasqure,

I would suggest you open an issue here: https://github.com/PaulStoffregen/cores

closed 2019-09-23 by sandeepmistry