arduino/Arduino #8549: v. 1.8.5 worked on my iMac. v. 1.8.8 crashes immediately at launch

RobLewis, 2019-02-16:
Here is the crash report: 

Process:               Arduino [516]
Path:                  /Applications/Arduino.app/Contents/MacOS/Arduino
Identifier:            cc.arduino.Arduino
Version:               1.8.8 (1.8.8)
Code Type:             X86-64 (Native)
Parent Process:        ??? [1]
Responsible:           Arduino [516]
User ID:               502

Date/Time:             2019-02-15 16:16:34.666 -0800
OS Version:            Mac OS X 10.11.6 (15G22010)
Report Version:        11
Anonymous UUID:        700FC5F4-AFFB-AEC2-2089-19D8307FAD0A

Time Awake Since Boot: 370 seconds

System Integrity Protection: enabled

Crashed Thread:        0  AppKit Thread  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
*** Terminating app due to uncaught exception 'NSInvalidArgumentException', reason: '-[NSApplicationAWT setAutomaticCustomizeTouchBarMenuItemEnabled:]: unrecognized selector sent to instance 0x7fbe23d49e40'
terminating with uncaught exception of type NSException
abort() called

Application Specific Backtrace 1:
0   CoreFoundation                      0x00007fff93d2a452 __exceptionPreprocess + 178
1   libobjc.A.dylib                     0x00007fff882a973c objc_exception_throw + 48
2   CoreFoundation                      0x00007fff93d9418d -[NSObject(NSObject) doesNotRecognizeSelector:] + 205
3   CoreFoundation                      0x00007fff93c9a4c1 ___forwarding___ + 1009
4   CoreFoundation                      0x00007fff93c9a048 _CF_forwarding_prep_0 + 120
5   libJTouchBar.dylib4973088345823033216 0x000000011bcf07c8 __JNI_OnLoad_block_invoke + 65
6   libdispatch.dylib                   0x00007fff890a993d _dispatch_call_block_and_release + 12
7   libdispatch.dylib                   0x00007fff8909e40b _dispatch_client_callout + 8
8   libdispatch.dylib                   0x00007fff890b1c1c _dispatch_main_queue_callback_4CF + 1685
9   CoreFoundation                      0x00007fff93cdf949 __CFRUNLOOP_IS_SERVICING_THE_MAIN_DISPATCH_QUEUE__ + 9
10  CoreFoundation                      0x00007fff93c9e82d __CFRunLoopRun + 1949
11  CoreFoundation                      0x00007fff93c9de28 CFRunLoopRunSpecific + 296
12  HIToolbox                           0x00007fff87d84935 RunCurrentEventLoopInMode + 235
13  HIToolbox                           0x00007fff87d84677 ReceiveNextEventCommon + 184
14  HIToolbox                           0x00007fff87d845af _BlockUntilNextEventMatchingListInModeWithFilter + 71
15  AppKit                              0x00007fff8cfc2df6 _DPSNextEvent + 1067
16  AppKit                              0x00007fff8cfc2226 -[NSApplication _nextEventMatchingEventMask:untilDate:inMode:dequeue:] + 454
17  libosxapp.dylib                     0x000000011951d328 -[NSApplicationAWT nextEventMatchingMask:untilDate:inMode:dequeue:] + 124
18  AppKit                              0x00007fff8cfb6d80 -[NSApplication run] + 682
19  libosxapp.dylib                     0x000000011951cf72 +[NSApplicationAWT runAWTLoopWithApp:] + 156
20  libsplashscreen.dylib               0x00000001194f95c8 __SplashInitPlatform_block_invoke_1 + 360
21  JavaNativeFoundation                0x000000011953dd60 +[JNFRunLoop _performCopiedBlock:] + 17
22  Foundation                          0x00007fff8eb1adbe __NSThreadPerformPerform + 279
23  CoreFoundation                      0x00007fff93cbf7e1 __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__ + 17
24  CoreFoundation                      0x00007fff93c9ef0c __CFRunLoopDoSources0 + 556
25  CoreFoundation                      0x00007fff93c9e42f __CFRunLoopRun + 927
26  CoreFoundation                      0x00007fff93c9de28 CFRunLoopRunSpecific + 296
27  libjli.dylib                        0x0000000104b328fe CreateExecutionEnvironment + 871
28  libjli.dylib                        0x0000000104b2e4cc JLI_Launch + 1952
29  Arduino                             0x00000001048c3aa3 launch + 14627
30  Arduino                             0x00000001048bff5f main + 175
31  Arduino                             0x00000001048bfea4 start + 52

Thread 0 Crashed:: AppKit Thread  Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff9213ff06 __pthread_kill + 10
1   libsystem_pthread.dylib       	0x00007fff9837f4ec pthread_kill + 90
2   libsystem_c.dylib             	0x00007fff8b13f6df abort + 129
3   libc++abi.dylib               	0x00007fff893a0b31 abort_message + 257
[... 1113 lines cut ...]
Memory Module: BANK 1/DIMM1, 2 GB, DDR2 SDRAM, 667 MHz, 0x7F7F7F7F7FF70000, 0x4F5743353343344444523253324742000000
AirPort: spairport_wireless_card_type_airport_extreme (0x14E4, 0x88), Broadcom BCM43xx 1.0 (5.10.131.39.16)
Bluetooth: Version 4.4.6f4, 3 services, 26 devices, 1 incoming serial ports
Network Service: Ethernet, Ethernet, en0
Network Service: Wi-Fi, AirPort, en1
Serial ATA Device: ST31000524AS, 1 TB
Parallel ATA Device: OPTIARC  DVD RW AD-5630A
USB Device: USB Bus
USB Device: USB Bus
USB Device: IR Receiver
USB Device: USB Bus
USB Device: USB Bus
USB Device: USB Bus
USB Device: Bluetooth HCI
USB Device: USB 2.0 Bus
USB Device: Keyboard Hub
USB Device: Kensington USB/PS2 Orbit
USB Device: FT232R USB UART
USB Device: Apple Keyboard
USB Device: AS2105
USB Device: USB 2.0 Bus
USB Device: Built-in iSight
USB Device: FANTOM DRIVE
FireWire Device: unknown_device, OWC Mercury Elite-AL Pro mini, Up to 800 Mb/sec
Thunderbolt Bus:

RobLewis, 2019-02-16:
I just reverted to v. 1.8.7 and it does not crash at launch.

per1234, 2019-02-16:
Closing as duplicate of https://github.com/arduino/Arduino/issues/8274.

Arduino IDE is not compatible with any macOS version older than 10.12. Please use Arduino IDE 1.8.7 until Arduino IDE 1.8.9 is released (which will not have the bug).

closed 2019-02-16 by per1234