arduino/Arduino #8553: mkr wifi 1010 pin layout

f0854, 2019-02-16:
I can't find the Arduino wifi 1010 pin layout, where is it?

per1234, 2019-02-16:
What do you mean by "pin layout"?

If you mean something that shows the mapping between the Arduino pin numbers and the pins on the SAMD21 microcontroller, there is one in the schematic on the "Documentation" tab of the product page:
https://store.arduino.cc/arduino-mkr-wifi-1010

However, I'd like to see Arduino add a "Pin Mapping" image to the product pages for the new boards, as was done for the classic Arduino boards:

---
[image]

f0854, 2019-02-16:
for pin layout I mean the description of the pins (see attached), however on the 1010 wifi the descriptions on the pcb of the pins that are above the card and those that are below are different
!    
[image]

per1234, 2019-02-16:
On the MKR WiFi 1010, there are no descriptions of the pins on the silkscreen on the top of the board. There are descriptions printed on sides of the female headers and on silkscreen on the bottom. Please list the differences you've found between the top of the PCB and the bottom of the PCB.

f0854, 2019-02-17:
I think there was a problem in assembling my mkr 1010, the descriptions printed on the sides of the female headers, on the left and right, of the card are "identical", so all the pins on the left are different from the silkscreen on the bottom  (see the attachments). Also there are pins numbered 1 to 7 that are? And the two connectors on the left and right side of the pcb are what? With my knowledge I can not use the card. I wonder: is it my fault if I can not find the documentation or so "open" the project is not and is deficient in the documentation?
Sorry for my late reply
[image]
[image]
[image]

f0854, 2019-02-17:
Sorry, the first picture is wrong, this is the right one
[image]

per1234, 2019-02-17:
You're correct that your MKR WiFi 1010 was incorrectly assembled. Each female header should have different and correct markings but yours uses the same header on both sides so only one of the headers has the correct markings. If this is a significant problem for you and you purchased this from the [Arduino Store](https://store.arduino.cc/), you can [contact Store Support ](https://store.arduino.cc/store-support)to see what they can do about resolving the issue. If you purchased the board from someone else, then you would need to contact them instead.

f0854, 2019-02-17:
thanks, I will contact Arduino's assistance ... nobody knows if there is official documentation somewhere? what do the numbers from zero to seven mean? What are the two side connectors and the SCL, SDA, MISO, SCK and MOSI pins that are? but above all, why is this documentation not on the arduino site?

closed 2019-03-05 by f0854