arduino/Arduino #8556: Does Programmers.local.txt ever get read?

NormanDunbar, 2019-02-18:
In Arduino IDE 1.8.5, on Windows 7 64bit, but as yet, untested on Linux.

I can create and edit `boards.local.txt`, in the same location as the installed `boards.txt`, and see my new boards on the tools->board menu when I next open the IDE. So far so good.

If I create a `programmers.local.txt`, in the same location as `programmers.txt`, I don't see my programmer(s) listed on the  tools->programmer menu. If I edit `programmers.txt` and add them there, they do appear on the menu.

Is this by design? Do I not have the option to add new programmers to a local version of the file, so that future IDE updates don't wipe out my ad-hoc programming devices?

Also, is there anywhere on the internmet that the contents of `programmers.txt` are documented like `boards.txt` and `platform.txt` are at https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5-3rd-party-Hardware-specification.

Cheers,
Norm.

NormanDunbar, 2020-04-13:
Afternoon All,
I realise you are busy with arduino-cli etc, but is there any chance of a response please? Can I create and use ``programmers.local.txt`` and is the ``programmers.txt` file documented anywhere?

Many thanks.

Cheers,
Norm.

per1234, 2020-08-26:
Hi @NormanDunbar. I know it's been a long time, but I have now submitted a pull request to fully document the programmers system of Arduino boards platforms. It's here: https://github.com/arduino/arduino-cli/pull/925
If you get a chance to take a look and see anything that is unclear or could be improved, please let me know.

As for your question about programmers.local.txt, this is not supported, but I think it would be nice to have, especially now that the available programmers have been limited to only those provided by the platform of the currently selected board (https://github.com/arduino/Arduino/pull/9900).

I hope https://github.com/arduino/arduino-cli/pull/925 will fulfill your request for documentation of the programmers system, but we can still leave this issue open to serve as the feature request for programmers.local.txt. Thanks!

NormanDunbar, 2020-08-26:
Hi @per1234 ,

Thanks very much. I'll have a look and see if there's anything I think needs adding/changing etc. Appreciate your time and effort. Thanks again.

Cheers,
Norm.