arduino/Arduino #8569: [Library manager] Please add this library

zerokol, 2019-02-22:
* eFLL (Embedded Fuzzy Logic Library) is a standard library for Embedded Systems

https://github.com/zerokol/eFLL

cmaglie, 2019-03-01:
Done

closed 2019-03-01 by cmaglie

zerokol, 2019-04-12:
@cmaglie how to update the library, I already created a new tag with the new value, I know that the crawler only runs in the Sunday, but it was weeks ago. Do I need to create a new issue?

per1234, 2019-04-12:
Releases will only be picked up by the indexer when they have a unique `version` value. Even though you updated it afterwards, at the time of your 1.1.1 release, the `version` value was still at 1.1.0:
https://github.com/zerokol/eFLL/blob/1.1.1/library.properties#L2

That is why your 1.1.1 release is not available in Library Manager.

So you need to either update the `version` value and create a new release, or else redo the existing 1.1.1 release with the correct `version` value this time.

More information:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

zerokol, 2019-04-13:
Oww My fault! Thank you very much!