arduino/Arduino #8576: [Library Manager] Please add the TFMini-Plus library to the Library Manager.

budryerson, 2019-02-25:
Please add this new addition to the Arduino Library Manager:
https://github.com/budryerson/TFMini-Plus

Despite its name, the Benewake TFmini-Plus sensor if completely different from the TFMini.  Currently, there is no Arduino library that supports this remarkable, inexpensive, time-of-flight, Lidar distance sensor.

This is my first effort at contributing to the Library Manager. Your gentle guidance will be appreciated.

per1234, 2019-02-25:
Hi. There are a couple of things that will need to be done before the repository is compliant with the requirements for inclusion in the Library Manager index:

Move library to the root of the repository. Currently the library is in the TFMPlus folder, which is not allowed.

After making that change, you need to create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing these things, please comment here to let us know we can proceed with adding your library to the Library Manager index.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

cmaglie, 2019-03-01:
Done

closed 2019-03-01 by cmaglie