arduino/Arduino #8577: Arduino IDE bug with Arduino Due

PeymanAmirii, 2019-02-26:
In Arduino IDE if you use this code to read from a sensor through I2C connection, It will works good for AVR based boards:

data=(Wire.read()|Wire.read()<<8 );

because the compiler will consider sign of MSB after 8bit shift (<<8) by paying attention to the sign bit of MSB. But when I used this code for Arduino Due, I found out that it will not consider the sign of MSB after 8 bit shift (<<8) because the compiler interpret all the bits as a number (will not pay attention to the sign bit) and it causes error in sensor readings. And now I remove this error for my code by a simpler way.
For more information, you can contact me.
peyman.amiri1@me.iut.ac.ir

PaulStoffregen, 2019-02-26:
This is a bug in your code, not Arduino.

Change it to:

data=(int16_t)(Wire.read()|Wire.read()<<8 );

Most of the motion sensor libraries with this sort of code were fixed years ago.

PeymanAmirii, 2019-02-27:
Thanks alot for your attention.

 

Links:
------
[1]
https://github.com/arduino/Arduino/issues/8577#issuecomment-467588532
[2]
https://github.com/

closed 2019-02-27 by per1234