arduino/Arduino #8589: raspberry pi3 hangs when compiling a sketch in arduino

Bringamosa, 2019-03-01:
It does not matter what sketch I try to compile, my pri will go into 100% CPU usage and hang. Even with the bare minimum sketch. 

Ant idea how I can start debugging

per1234, 2019-03-01:
Which version of the Arduino IDE are you using?

Bringamosa, 2019-03-02:
It says Arduino 1.8.8
I'm running raspbian stretch, updated and upgraded. 
The only thing else running on the arduino are the vnc service and thanks iot server with the postgres database. 

I am fairly new to the arduino/linux world so not sure if there is any kind of build in logging that can show what's going on when the vncscreen/IDE freezes.

Once frozen the rpi will stay on the wifi but I can not reach or via vnc anymore

gdsports, 2019-04-17:
When running the GUI do you see a yellow lightning bolt on the top right? This indicates a bad power supply which might cause freezes when the CPU is under heavy load.

https://raspberrypi.stackexchange.com/questions/57963/what-does-the-lightning-bolt-mean

tariqshiwani, 2019-08-21:
i am also having same issue, tried to compile blink sketch from examples.
IDE 1.8.9
raspberry Pi 3 32bit
no lightning bolt

works fine in 1.8.5 but it takes around 5 minutes to compile that blink sketch

gdsports, 2019-08-22:
What is the board type? Uno, ESP8266, etc.

tariqshiwani, 2019-08-22:
NodeMCU, ESP8266

Bringamosa, 2019-08-22:
I think I read somewhere that it's some issue with the arduino IDE using up too much resources

tariqshiwani, 2019-08-22:
i saw your other thread about this issue and some people have replied to it
as well but some of those responses are above my head :)

1.8.5 works fine, the parallel build feature was introduced since 1.8.6 and
I guess that's where the problem begins.

I saw the release note of 1.8.9 which says "New target! ARM64 boards are
now fully supported (Nvidia Jetson and RaspberryPi3 with 64bit operating
system)" not sure if it is referring to the same issue but even if it is,
it's about 64bit version of Pi not 32 (the one I have)

gdsports, 2019-08-23:
I verified an RPi 4 with 4GB of RAM and Arduino LinuxARM 32-bit will compile the blink sketch for board type ESP8266 without freezing. 4GB RAM versus 1GB avoids the out of memory problems. Running full Raspbian up to date and VNC.

tariqshiwani, 2019-08-23:
ok but that's Pi4, in my case I have Pi3 model B which has 1.2g quad core
with 1gb RAM which unfortunately can't be increased, so I guess I am out of
luck then :(

facchinm, 2019-08-23:
Hi all,
the IDE (or, better, the builder) compiles using `N = cores` threads (see https://github.com/arduino/arduino-builder/blob/d9d12abc0fa2285bb4debddab683eadb158aa47b/arduino-builder/main.go#L169 ).
When we call it from Java IDE there's no override of the default setting (https://github.com/arduino/Arduino/blob/b5bfe08bd2c6f45d6762fad4cc9337063406d92a/arduino-core/src/cc/arduino/Compiler.java#L233) so the 4-cores Rpi3 will try compiling 4 files at the same time and run out of memory (on some "heavy" cores like ESP8266/32)
My suggestion is to provide a way to override that value via a platform.txt flag, not exposed by the Preferences UI. Could it be ok for everyone?

gdsports, 2019-08-23:
The flag has no effect on the number of copies of gcc/g++ running at the same time. I have seen as many as 54 copies of cc1plus running at the same time. On systems with 1GB of RAM, physical RAM is quickly exhausted so the OS thrashes the swap file.

The following is from a terminal window while compiling the blink sketch from scratch (build all) for a Generic ESP8266.

```
pi@rpi44gb:~/arduino-1.8.9 $ ps aux|grep cc1plus |wc
      2      94    1582
pi@rpi44gb:~/arduino-1.8.9 $ ps aux|grep cc1plus |wc
      1      13      91
pi@rpi44gb:~/arduino-1.8.9 $ ps aux|grep cc1plus |wc
     54    4306   82217
pi@rpi44gb:~/arduino-1.8.9 $ ps aux|grep cc1plus |wc
     43    3415   65239
pi@rpi44gb:~/arduino-1.8.9 $ ps aux|grep cc1plus |wc
     40    3172   60540
pi@rpi44gb:~/arduino-1.8.9 $ ps aux|grep cc1plus |wc
     39    3091   58964
pi@rpi44gb:~/arduino-1.8.9 $ ps aux|grep cc1plus |wc
     38    3010   57442
pi@rpi44gb:~/arduino-1.8.9 $ ps aux|grep cc1plus |wc
     36    2848   54317
pi@rpi44gb:~/arduino-1.8.9 $ ps aux|grep cc1plus |wc
     36    2848   54317
pi@rpi44gb:~/arduino-1.8.9 $ ps aux|grep cc1plus |wc
     33    2605   49668
pi@rpi44gb:~/arduino-1.8.9 $ ps aux|grep cc1plus |wc
     26    2038   38854
pi@rpi44gb:~/arduino-1.8.9 $ ps aux|grep cc1plus |wc
     12     904   17121
pi@rpi44gb:~/arduino-1.8.9 $ ps aux|grep cc1plus |wc
      3     175    3140
pi@rpi44gb:~/arduino-1.8.9 $ ps aux|grep cc1plus |wc

```

gdsports, 2019-08-23:
arduino-cli also spawns off far more than 4 gcc/g++ processes.

`$ arduino-cli compile --fqbn esp8266:esp8266:generic Arduino/blink/blink.ino `

In a different terminal window,
```
pi@rpi44gb:~ $ ps aux|grep cc1 |wc
     39    3091   63178
pi@rpi44gb:~ $ ps aux|grep cc1 |wc
     37    2929   59879
pi@rpi44gb:~ $ ps aux|grep cc1 |wc
     37    2929   59879
pi@rpi44gb:~ $ ps aux|grep cc1 |wc
     37    2929   59879
pi@rpi44gb:~ $ ps aux|grep cc1 |wc
     31    2443   49870
pi@rpi44gb:~ $ ps aux|grep cc1 |wc
     23    1795   36651
pi@rpi44gb:~ $ ps aux|grep cc1 |wc
     12     904   18263
pi@rpi44gb:~ $ ps aux|grep cc1 |wc
      5     337    6652
```

tariqshiwani, 2019-08-24:
I read the same somewhere, but not sure how to get it fixed.

facchinm, 2019-08-26:
@gdsports thanks for checking, I tested it and indeed the builder logic to limit concurrent processes is flawed. I prepared a patch here https://github.com/arduino/arduino-builder/compare/master...facchinm:fix_fake_parallel_limitation?expand=1 and here https://github.com/arduino/arduino-cli/pull/356 ; this should really limit the maximum concurrent processes to N=cores number, so maybe we don't need the extra `-j1` patch for the IDE

tariqshiwani, 2019-08-26:
great @facchinm
but how do we get the new build?
do we have to download the complete source code and build ourselves?

facchinm, 2019-08-26:
I'll push the builder PR to get the prebuilt binary for testing ;)

facchinm, 2019-08-26:
@tariqshiwani you can grab the build here https://github.com/arduino/arduino-builder/pull/331#issuecomment-524775907

gdsports, 2019-08-26:
@facchinm Thanks, the patched arduino-builder works fine. I do not think the "-j1" patch is needed. The GUI performance is excellent even while running parallel compiles.

tariqshiwani, 2019-08-26:
@facchinm  thanks for the patch but it didn't work for me.

i am getting following error when hit compile on Blink sample sketch

Arduino: 1.8.9 (Linux), Board: "NodeMCU 1.0 (ESP-12E Module), 80 MHz, Flash, Disabled, All SSL ciphers (most compatible), 4M (no SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 115200"

/home/pi/arduino/arduino-1.8.9/arduino-builder: 1: /home/pi/arduino/arduino-1.8.9/arduino-builder: Syntax error: "(" unexpected
/home/pi/arduino/arduino-1.8.9/arduino-builder returned 2

Error compiling for board NodeMCU 1.0 (ESP-12E Module).

tariqshiwani, 2019-08-26:
Sorry my bad, please disregard my previous comments, I mistakenly copied linux32 instead of linuxarm

its working perfectly fine. 

Thank you guys @Bringamosa @gdsports @gdsports

cmaglie, 2019-09-10:
The patch to the builder has been merged and scheduled for the next IDE release.

closed 2019-09-10 by cmaglie

wtarreau, 2020-09-06:
Hi, just a quick report to confirm that I faced the exact same issue with 1.8.9 on linux32 (with 54 cc1plus processes as well), and that upgrading to 1.8.13 solved the problem. Thank you!