arduino/Arduino #8597: [Library manager] Not delete files that no longer in the repository.

JoaoLopesF, 2019-03-03:
Hi, I migrate the sources of my library, to a "src" folder.
When update that library by Library Manager, the files that were moved,
not is deleted, so the sources files will be in 2 locations.
Can change the Library manager to delete files that no longer exists in github repo ?

Regards

per1234, 2019-03-04:
Please provide a link to the library in question and tell us which release the source files migration was done in so we can use that for testing.

JoaoLopesF, 2019-03-20:
Hi @cmaglie 

To reproduce:

Use the library manager, and download RemoteDebug v 1.5.9
see the files in library folder.
Use the library manager, and download RemoteDebug v 3.0.4
see the files in library folder. the cpp and h files is in root of this folder and in src directory

per1234, 2019-03-21:
@JoaoLopesF I just tried following your instructions a few times but was unable to reproduce the issue. I'm using Arduino IDE 1.8.9 on Windows 10.

If you have not done so already, please try it with Arduino IDE 1.8.9 to make sure the issue hasn't already been fixed and let us know what your results are.

Also, which operating system are you using?

sandeepmistry, 2019-09-16:
Based on @per1234's comment above, I'm closing this for now due to lack of feedback. Please re-open with requested info if you are still interested.

closed 2019-09-16 by sandeepmistry