arduino/Arduino #8601: Add "IoTtweetSIEMENS_SIMATIC" library to Library manager Arduino IDE

Isaranu, 2019-03-05:
We are developer team from www.iottweet.com which provided an IoT dashboard using for "SIEMENS SIMATIC IOT2000 series" send data to display. It's compatible use with NB-IoT device (provider : AIS)

github url : https://github.com/Isaranu/IoTtweetSIEMENS_SIMATIC

Thank you very much.
IoTtweet team.

per1234, 2019-03-05:
There is a change you need to make before your library can be added to the Library Manager index:

Rename the file `libraries.properties` to `library.properties`.

After doing that, you will need to either create a new tag/release or else redo the existing tag/release. If you are creating a new tag, please make sure to update the `version` value in library.properties before creating the tag/release.

After doing that, please comment here so we'll know we can proceed.

More information:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#library-metadata

Isaranu, 2019-03-06:
> There is a change you need to make before your library can be added to the Library Manager index:
>
> Rename the file `libraries.properties` to `library.properties`.
>
> After doing that, you will need to [...]

Hi, 
Thank you very much for your supported.
I do renamed a library.properties file. And change version to v.1.1 (both in library.properties file and release new tag version)
please check and do next proceed.

Thank you so much

cmaglie, 2019-03-11:
Done

closed 2019-03-11 by cmaglie