arduino/Arduino #8603: High CPU load. 1.9b Mac OS X 10.11.6

danbates2, 2019-03-05:
I had all windows of the program closed, it was idling in the background.
Mac OSX10.11.6
Sample Process:
```
Sampling process 73641 for 3 seconds with 1 millisecond of run time between samples
Sampling completed, processing symbols...
Analysis of sampling Arduino (pid 73641) every 1 millisecond
Process:         Arduino [73641]
Path:            /Applications/Arduino.app/Contents/MacOS/Arduino
Load Address:    0x100fbc000
Identifier:      cc.arduino.Arduino
Version:         PR-beta1.9-BUILD-102 (1.8.9)
Code Type:       X86-64
Parent Process:  ??? [1]

Date/Time:       2019-03-05 16:28:10.907 +0000
Launch Time:     2019-03-05 11:07:45.127 +0000
OS Version:      Mac OS X 10.11.6 (15G22010)
Report Version:  7
Analysis Tool:   /usr/bin/sample
----

Call graph:
    1504 Thread_2159227   DispatchQueue_1: com.apple.main-thread  (serial)
    + 1504 start  (in Arduino) + 52  [0x100fbcea4]
    +   1504 main  (in Arduino) + 175  [0x100fbcf5f]
    +     1504 launch  (in Arduino) + 14627  [0x100fc0aa3]
    +       1504 JLI_Launch  (in libjli.dylib) + 1952  [0x1012284cc]
    +         1504 CreateExecutionEnvironment  (in libjli.dylib) + 871  [0x10122c8fe]
    +           1504 CFRunLoopRunSpecific  (in CoreFoundation) + 296  [0x7fff966e7e28]
    +             1504 __CFRunLoopRun  (in CoreFoundation) + 927  [0x7fff966e842f]
    +               1504 __CFRunLoopDoSources0  (in CoreFoundation) + 556  [0x7fff966e8f0c]
    +                 1504 __CFRUNLOOP_IS_CALLING_OUT_TO_A_SOURCE0_PERFORM_FUNCTION__  (in CoreFoundation) + 17  [0x7fff967097e1]
    +                   1504 __NSThreadPerformPerform  (in Foundation) + 279  [0x7fff9d285dbe]
    +                     1504 +[JNFRunLoop _performCopiedBlock:]  (in JavaNativeFoundation) + 20  [0x117b8f5f5]
    +                       1504 __SplashInitPlatform_block_invoke_1  (in libsplashscreen.dylib) + 360  [0x117b4b5c8]
    +                         1504 +[NSApplicationAWT runAWTLoopWithApp:]  (in libosxapp.dylib) + 156  [0x117b6ef72]
    +                           1504 -[NSApplication run]  (in AppKit) + 796  [0x7fff98631df2]
    +                             1504 -[NSApplicationAWT sendEvent:]  (in libosxapp.dylib) + 378  [0x117b6f2aa]
    +                               1504 -[NSApplication sendEvent:]  (in AppKit) + 4274  [0x7fff987cb0fe]
    +                                 1504 -[NSApplication _handleKeyEquivalent:]  (in AppKit) + 920  [0x7fff988a4949]
    +                                   1504 -[NSMenu performKeyEquivalent:]  (in AppKit) + 357  [0x7fff988a57a5]
    +                                     1504 _os_activity_initiate  (in libsystem_trace.dylib) + 75  [0x7fff8ee6307a]
    +                                       1504 -[NSCarbonMenuImpl performActionWithHighlightingForItemAtIndex:]  (in AppKit) + 114  [0x7fff988a68b7]
    +                                         1504 -[NSMenuItem _corePerformAction]  (in AppKit) + 336  [0x7fff988a6b57]
    +                                           1504 -[NSApplication sendAction:to:from:]  (in AppKit) + 460  [0x7fff988a6dbd]
    +                                             1504 _os_activity_initiate  (in libsystem_trace.dylib) + 75  [0x7fff8ee6307a]
    +                                               1504 -[NSApplication terminate:]  (in AppKit) + 762  [0x7fff9885366d]
    +                                                 1504 -[NSApplication _shouldTerminate]  (in AppKit) + 1084  [0x7fff98854092]
    +                                                   1504 -[NSApplicationAWT sendEvent:]  (in libosxapp.dylib) + 378  [0x117b6f2aa]
    +                                                     1504 -[NSApplication sendEvent:]  (in AppKit) + 1841  [0x7fff987ca77d]
    +                                                       1504 -[AWTWindow_Normal sendEvent:]  (in libawt_lwawt.dylib) + 86  [0x118fed00b]
    +                                                         1504 -[NSWindow sendEvent:]  (in AppKit) + 517  [0x7fff9884a539]
    +                                                           1504 -[NSWindow _reallySendEvent:isDelayedEvent:]  (in AppKit) + 2108  [0x7fff98e0bb15]
    +                                                             1439 -[AWTView deliverJavaKeyEventHelper:]  (in libawt_lwawt.dylib) + 310  [0x118feb331]
    +                                                             ! 1439 JNFNewObject  (in JavaNativeFoundation) + 174  [0x117b8c302]
    +                                                             !   1439 jni_NewObjectV  (in libjvm.dylib) + 61  [0x10402979f]
    +                                                             !     1439 ThreadStateTransition::trans_from_native(JavaThreadState)  (in libjvm.dylib) + 150  [0x103e7d752]
    +                                                             !       1439 JavaThread::check_safepoint_and_suspend_for_native_trans(JavaThread*)  (in libjvm.dylib) + 232  [0x10428171e]
    +                                                             !         1439 SafepointSynchronize::block(JavaThread*)  (in libjvm.dylib) + 375  [0x1041f83fb]
    +                                                             !           1439 Monitor::lock_without_safepoint_check()  (in libjvm.dylib) + 39  [0x10417a301]
    +                                                             !             1134 Monitor::ILock(Thread*)  (in libjvm.dylib) + 310  [0x10417a284]
    +                                                             !             : 1134 ParkCommon(ParkEvent*, long)  (in libjvm.dylib) + 42  [0x104179bc2]
    +                                                             !             :   1134 os::PlatformEvent::park()  (in libjvm.dylib) + 192  [0x10419cdde]
    +                                                             !             :     1134 _pthread_cond_wait  (in libsystem_pthread.dylib) + 767  [0x7fff8fcc3728]
    +                                                             !             :       1134 __psynch_cvwait  (in libsystem_kernel.dylib) + 10  [0x7fff9f327db6]
    +                                                             !             305 Monitor::ILock(Thread*)  (in libjvm.dylib) + 328  [0x10417a296]
    +                                                             !               305 ParkCommon(ParkEvent*, long)  (in libjvm.dylib) + 42  [0x104179bc2]
    +                                                             !                 305 os::PlatformEvent::park()  (in libjvm.dylib) + 192  [0x10419cdde]
    +                                                             !                   305 _pthread_cond_wait  (in libsystem_pthread.dylib) + 767  [0x7fff8fcc3728]
[... 1420 lines cut ...]
    +         ! : | + 16 void PSParallelCompact::mark_and_push<unsigned int>(ParCompactionManager*, unsigned int*)  (in libjvm.dylib) + 127  [0x103fd7ef1]
    +         ! : | + ! 16 ParMarkBitMap::mark_obj(HeapWord*, unsigned long)  (in libjvm.dylib) + 181,26,...  [0x1041adbc7,0x1041adb2c,...]
    +         ! : | + 4 void PSParallelCompact::mark_and_push<unsigned int>(ParCompactionManager*, unsigned int*)  (in libjvm.dylib) + 149  [0x103fd7f07]
    +         ! : | + ! 4 ParallelCompactData::add_obj(HeapWord*, unsigned long)  (in libjvm.dylib) + 100,67,...  [0x1041d7f4c,0x1041d7f2b,...]
    +         ! : | + 3 void PSParallelCompact::mark_and_push<unsigned int>(ParCompactionManager*, unsigned int*)  (in libjvm.dylib) + 106  [0x103fd7edc]
    +         ! : | +   3 oopDesc::size()  (in libjvm.dylib) + 15,48,...  [0x103ebdd6f,0x103ebdd90,...]
    +         ! : | 20 ObjArrayKlass::oop_follow_contents(ParCompactionManager*, oopDesc*)  (in libjvm.dylib) + 112  [0x10418f988]
    +         ! : | + 20 void ObjArrayKlass::objarray_follow_contents<unsigned int>(ParCompactionManager*, oopDesc*, int)  (in libjvm.dylib) + 102  [0x1041905e4]
    +         ! : | +   9 void PSParallelCompact::mark_and_push<unsigned int>(ParCompactionManager*, unsigned int*)  (in libjvm.dylib) + 127  [0x103fd7ef1]
    +         ! : | +   ! 9 ParMarkBitMap::mark_obj(HeapWord*, unsigned long)  (in libjvm.dylib) + 241,48,...  [0x1041adc03,0x1041adb42,...]
    +         ! : | +   5 void PSParallelCompact::mark_and_push<unsigned int>(ParCompactionManager*, unsigned int*)  (in libjvm.dylib) + 122,9,...  [0x103fd7eec,0x103fd7e7b,...]
    +         ! : | +   4 void PSParallelCompact::mark_and_push<unsigned int>(ParCompactionManager*, unsigned int*)  (in libjvm.dylib) + 106  [0x103fd7edc]
[... cut ...]
    +         ! : | + 3 InstanceKlass::vtable_length() const  (in libjvm.dylib) + 10,0,...  [0x103fc95bc,0x103fc95b2,...]
    +         ! : | 1 InstanceKlass::oop_follow_contents(ParCompactionManager*, oopDesc*)  (in libjvm.dylib) + 93  [0x103fd4b71]
    +         ! : | + 1 void PSParallelCompact::mark_and_push<oopDesc*>(ParCompactionManager*, oopDesc**)  (in libjvm.dylib) + 6  [0x103fd7f68]
    +         ! : | 1 InstanceMirrorKlass::oop_follow_contents(ParCompactionManager*, oopDesc*)  (in libjvm.dylib) + 32  [0x103fe14e2]
    +         ! : | + 1 InstanceKlass::

facchinm, 2019-03-06:
HI @danbates2 ,
first of all thanks for reporting! Is the issue reproducible without any effort (like, do you just need to open the ide and close it to trigger the high cpu occupation)? Or do you need extra steps? From the trace, it looks like all threads are waiting so cpu occupation should be close to 0%

danbates2, 2019-03-06:
Hi @facchinm 
It's not easily reproducible.
It's possible it finished it's high CPU load before I took the snapshot.
It seems to idle with no windows open at around 3%.

arduino-builder process was present in my task manager while Arduino IDE was closed. Is this normal?

facchinm, 2019-03-06:
> arduino-builder process was present in my task manager while Arduino IDE was closed. Is this normal?

No, unless you were compiling something. Recent versions of the IDE don't kill the app when the last editor gets closed, so all children processes are still alive (and the builder can very likely reach 650% cpu usage on an 8 core machine since it compiles parallely).

I'd close the issue as "temporary side effect", the important thing is that it stops taking cpu time sooner or later. Feel free to reopen if it happens again, thanks!

closed 2019-03-06 by facchinm