arduino/Arduino #8609: [Library] New library ClosedCube STS35

closedcube, 2019-03-06:
Hello,

Please add new library: 
https://github.com/closedcube/ClosedCube_STS35_Arduino

Thanks and regards,
ClosedCube

per1234, 2019-03-07:
Hi. Before we can add your library to the Library Manager index, there is an issue that must be resolved:

The `version` value in [your library.properties](https://github.com/closedcube/ClosedCube_STS35_Arduino/blob/2019.03.06.1/library.properties#L2) file is not compliant with the Arduino library specification:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format
>version - version of the library. Version should be [semver](http://semver.org/) compliant. 1.2.0 is correct; 1.2 is accepted; r5, 003, 1.1c are invalid

After correcting the `version` value, you will need to either create a new tag/release or else redo the current tag/release.

Once that's done, please comment here so we will know we can proceed.

closedcube, 2019-03-07:
HI,

_version_ has been updated.

Many thanks

cmaglie, 2019-03-11:
Done

closed 2019-03-11 by cmaglie