arduino/Arduino #8610: [Library] New library ClosedCube I2C Driver (Wrapper)

closedcube, 2019-03-06:
Hello,

Please add new library: 

https://github.com/closedcube/ClosedCube_I2C_Arduino

Thanks and regards,
ClosedCube

per1234, 2019-03-07:
Note that, as with https://github.com/arduino/Arduino/issues/8609, the `version` value in your library.properties file is not compliant with the [Arduino library specification](https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification). However, this repository does have previous releases that are compliant so I don't think the issue will block addition to the Library Manager index in this case. It will mean that the latest release of the library (and any other non-compliant releases that come after it) will not be available for installation via Library Manager.

closedcube, 2019-03-07:
HI,

_version_ has been updated.

Many thanks

cmaglie, 2019-03-11:
It seems that you renamed the library from `ClosedCube I2C driver` to `ClosedCube_I2C_Driver`, shall I keep the latter?

closedcube, 2019-03-12:
Yes, please

Thanks

cmaglie, 2019-03-18:
done

closed 2019-03-18 by cmaglie