arduino/Arduino #8614: [Library Manager] Add ZumoAutomation to Library Manager

jsmith2021Brandeis, 2019-03-07:
I have prepared the following repository to be added to the Arduino Library Manager
Zumo Automation
Url: https://github.com/BrandeisMakerLab/Zumo-Automation

It allows the Pololu Zumo Shield robot to play Zumo matches with calibrated line sensors, timers, and driving functions

per1234, 2019-03-07:
Hi. Although the repository is now compliant with the requirements for inclusion in the Library Manager index, it was not compliant at the time of the last tag release.

Please do the following:

1. Update the `version` field in library.properties to the version number you will use for the new release.
2. Create a [release](https://help.github.com/articles/creating-releases/) or [tag](https://git-scm.com/docs/git-tag) that matches the `version` value in your library.properties file.

After doing so, please comment here to let us know we can proceed with adding your library to the Library Manager index.

jsmith2021Brandeis, 2019-03-07:
Ok, I changed version to 1.0.1 and made a new release

cmaglie, 2019-03-11:
Done

closed 2019-03-11 by cmaglie