arduino/Arduino #8617: Arduino IDE Crashed Soon After "ant run"(WIN7)

cuter1986, 2019-03-07:
According to this guide: [https://github.com/arduino/Arduino/wiki/Building-Arduino]
I succeeded in compiling arduino ide source code, but when I try to launch it using "ant run", it immediately crashes.  What can i do to help?

My OS is WIN7.
jdk installation path: D:\Program Files (x86)\Java\jdk1.8.0_202\
JAVA_HOME = D:\Program Files (x86)\Java\jdk1.8.0_202\
CLASSPATH = .%JAVA_HOME%\lib;%JAVA_HOME%\lib\tools.jar

arduino_debug.exe reports following info.

D:\Arduino\build\windows\work>arduino_debug.exe
java.lang.NoClassDefFoundError: Could not initialize class com.sun.jna.Native
        at com.sun.jna.platform.win32.Shell32.<clinit>(Shell32.java:32)
        at com.sun.jna.platform.win32.Shell32Util.getFolderPath(Shell32Util.java
:44)
        at com.sun.jna.platform.win32.Shell32Util.getFolderPath(Shell32Util.java
:61)
        at cc.arduino.os.windows.Win32KnownFolders.getLocalAppDataFolder(Win32Kn
ownFolders.java:53)
        at processing.app.windows.Platform.recoverSettingsFolderPath(Platform.ja
va:64)
        at processing.app.windows.Platform.init(Platform.java:54)
        at processing.app.Base.<init>(Base.java:208)
        at processing.app.Base.main(Base.java:151)

facchinm, 2019-03-08:
Hi @cuter1986 ,
I think `ant run` also needs `.../Arduino/build/windows/work/lib/` to be in the classpath. Can you check if it doesn't crash after adding that path?

cuter1986, 2019-03-11:
I add .../Arduino/build/windows/work/lib/ to CLASSPATH. still the same problem.

cuter1986, 2019-03-12:
I just copy the compiled files to another computer(computer B), arduino.exe run successfully.
there must be some wrong with my running environment. I'll build source on computer B and see if everything is fine.

Pieter12345, 2019-03-15:
> 
> 
> Hi @cuter1986 ,
> I think `ant run` also needs `.../Arduino/build/windows/work/lib/` to be in the classpath. Can you check if it doesn't crash after adding that path?
Adding `...\Arduino\build\windows\work\lib` to the `Path` system variable does not affect the behavior of executing `arduino.exe` or `arduino_debug.exe` inside the `Arduino/build/windows/work` directory (it still gives the same exception).

Pieter12345, 2019-03-15:
Update:
I've copied the `C:\Users\X\Documents\Java Eclipse Workspace\Arduino` directory to `C:\Users\X\Documents\Java Eclipse Workspace2\Arduino` and then executed `C:\Users\X\Documents\Java Eclipse Workspace2\Arduino\build\windows\work\arduino.exe` and the original `C:\Users\X\Documents\Java Eclipse Workspace\Arduino\build\windows\work\arduino.exe`. This results in the copy starting perfectly fine and the original file still crashing.
I have closed Eclipse and removed the `work/lib` directory from my path to validate that this doesn't influence it. Using Process Explorer to detect processes that use / contain access to the `C:\Users\X\Documents\Java Eclipse Workspace\Arduino\build\windows\work` directory resulted in 0 matches.

Pieter12345, 2019-03-15:
[image]
Apparentally the directories inside the `work` directories are shared with over 300 share names (e.g. R05_Inputs, R07_Runaway_Robot, R08_Remote_Control, Beep, GCC, RTC, ...) that require a confirmation per share name to be removed.
After removing the entire `work` directory and replacing it by the copied version of it, `arduino.exe` runs fine.
Why are directories shared under hundreds of names? Is this necessary for the build process?

facchinm, 2019-03-18:
Mmmmh this is something totally unexpected, maybe just a side effect of an operation we are doing during build phase. Also, it's very strange that share names affects classpath... I'm taking a look more closely on a Win10 installation, but we actually run all the tests (via Jenkins) on a real windows machine and I've never seen something like this.

Pieter12345, 2019-03-18:
@facchinm If you fail to reproduce this issue, feel free to ask me to perform some tests. I'm running Windows 7 64 bits with exactly the same 32 bits java version as the author of this issue.

facchinm, 2019-03-18:
I was unable to reproduce with a clean installation of Windows 10 + JDK 8u201 (32 bit) , so the problem must be lying somewhere else (maybe in the tool used to clone the repo? I went directly through .zip download)

Pieter12345, 2019-03-18:
@facchinm The problem occurs when building Arduino using "ant dist". The generated `windows/work` directory has all these 'shared' files/directories in it. When this generated `windows/work` directory is copied to another location on the disk, it runs fine.

Pieter12345, 2019-03-19:
After more testing, I found out that `jna-4.2.2.jar` is not properly loading due to the `jnidispatch-4.2.2-win32-x86.dll` file containing very odd user permissions.

Right clicking `jnidispatch-4.2.2-win32-x86.dll` -> Properties and selecting the `Security` tab results in the following warning:
[image]

The user permissions for a copy of `jnidispatch-4.2.2-win32-x86.dll` (left) and the unchanged `jnidispatch-4.2.2-win32-x86.dll` (right) show as:
[image]

When clicking `Edit`, the following dialog opens:
[image]

However, clicking `Reorder` does not change anything (fails seemlessly).

MeAmAnUsername, 2019-03-19:
I have the same issue on windows 10
I also have the weird security permissions after `ant build`.
Moving `jnidispatch-4.2.2-win32-x86.dll` and then copying it back fixes the issue.
Moving it and then moving it back does not fix the issue. This suggests that some of the properties of that file are incorrect, and they get fixed when copying the file.

I did some testing, and it seems to be caused by the cygwin `unzip` executable. Manually unzipping with that executable gives the weird security permissions (and a run failure for `arduino.exe`), manually running the Git `unzip` gives normal security permissions without issues.

I also tried substituting the cygwin unzip with the git `unzip` to see if `ant` would build it correctly but instead `ant` fails with the following error:
```
unzip:
     [echo] Unzipping windows/avr-gcc-5.4.0-atmel3.6.1-arduino2-i686-w64-mingw32.zip into folder windows/work/hardware/tmp/gcc/

BUILD FAILED
path\to\Arduino\build\build.xml:124: The following error occurred while executing this line:
path\to\Arduino\build\build.xml:118: The following error occurred while executing this line:
path\to\Arduino\build\build.xml:1046: The following error occurred while executing this line:
path\to\Arduino\build\build.xml:1179: The following error occurred while executing this line:
path\to\Arduino\build\build.xml:1256: The following error occurred while executing this line:
path\to\Arduino\build\build.xml:908: exec returned: -1073741515
```
The exit code is the windows exit code for a missing `.dll`

I also tried `ant run` from git bash instead of cygwin, and that worked (IDE starts successfully).

cuter1986, 2019-03-20:
I tried "Moving jnidispatch-4.2.2-win32-x86.dll and then copying it back". 
After fix the  security permissions issue of jnidispatch-4.2.2-win32-x86.dll, directly run arduino.exe, it starts successfully. `ant run` will not work, cause `ant run` will regenerate the dll file.

facchinm, 2019-03-20:
I've been testing using Git on Windows infrastructure and not cygwin, this is why I could reproduce the issue... Thanks everyone for the incredible collaborative effort, if #8682 builds fine I'll immediately merge it.

PR #8682 (2019-04-18): Replace "unzip/tar" in build.xml by Ant's provided unzip/untar method
Replace the "unzip" and "tar" command used while building Arduino by the unzip and untar method that Ant provides.
This fixes #8617, meaning that "ant run" will work again and no files/directories are created with weird permissions and share names anymore.
commit 126c79a3 (2019-04-18): Replace "unzip/tar" in build.xml by Ant's provided unzip/untar method

Replace the "unzip" and "tar" command used while building Arduino by the unzip and untar method that Ant provides.
This fixes #8617, meaning that "ant run" will work again and no files/directories are created with weird permissions and share names anymore.
closed 2019-04-18 by facchinm