arduino/Arduino #8622: Error compiling for board-> xtensa-lx106-elf-gcc: fatal error: -fuse-linker-plugin, but liblto_plugin.so not found

mmphego, 2019-03-08:
Hi,
I am having issues compiling for most boards including esp8266

Getting this error: **xtensa-lx106-elf-gcc: fatal error: -fuse-linker-plugin, but liblto_plugin.so not found**

------------------------------------------------------------------------------------

$ ./arduino
```shell
Picked up JAVA_TOOL_OPTIONS: 
 
/home/mmphego/Apts/arduino-1.8.8/arduino-builder -dump-prefs -logger=machine -hardware /home/mmphego/Apts/arduino-1.8.8/hardware -hardware /home/mmphego/.arduino15/packages -tools /home/mmphego/Apts/arduino-1.8.8/tools-builder -tools /home/mmphego/Apts/arduino-1.8.8/hardware/tools/avr -tools /home/mmphego/.arduino15/packages -built-in-libraries /home/mmphego/Apts/arduino-1.8.8/libraries -libraries /home/mmphego/Arduino/libraries -fqbn=esp8266:esp8266:d1:CpuFrequency=80,VTable=flash,FlashSize=4M1M,LwIPVariant=v2mss536,Debug=Disabled,DebugLevel=None____,FlashErase=none,UploadSpeed=921600 -ide-version=10808 -build-path /tmp/arduino_build_157338 -warnings=all -build-cache /tmp/arduino_cache_128751 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool.path=/home/mmphego/.arduino15/packages/esp8266/tools/esptool/0.4.13 -prefs=runtime.tools.esptool-0.4.13.path=/home/mmphego/.arduino15/packages/esp8266/tools/esptool/0.4.13 -prefs=runtime.tools.mkspiffs.path=/home/mmphego/.arduino15/packages/esp8266/tools/mkspiffs/0.2.0 -prefs=runtime.tools.mkspiffs-0.2.0.path=/home/mmphego/.arduino15/packages/esp8266/tools/mkspiffs/0.2.0 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=/home/mmphego/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2 -prefs=runtime.tools.xtensa-lx106-elf-gcc-1.20.0-26-gb404fb9-2.path=/home/mmphego/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2 -verbose /tmp/untitled96212026.tmp/sketch_mar08a/sketch_mar08a.ino
/home/mmphego/Apts/arduino-1.8.8/arduino-builder -compile -logger=machine -hardware /home/mmphego/Apts/arduino-1.8.8/hardware -hardware /home/mmphego/.arduino15/packages -tools /home/mmphego/Apts/arduino-1.8.8/tools-builder -tools /home/mmphego/Apts/arduino-1.8.8/hardware/tools/avr -tools /home/mmphego/.arduino15/packages -built-in-libraries /home/mmphego/Apts/arduino-1.8.8/libraries -libraries /home/mmphego/Arduino/libraries -fqbn=esp8266:esp8266:d1:CpuFrequency=80,VTable=flash,FlashSize=4M1M,LwIPVariant=v2mss536,Debug=Disabled,DebugLevel=None____,FlashErase=none,UploadSpeed=921600 -ide-version=10808 -build-path /tmp/arduino_build_157338 -warnings=all -build-cache /tmp/arduino_cache_128751 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool.path=/home/mmphego/.arduino15/packages/esp8266/tools/esptool/0.4.13 -prefs=runtime.tools.esptool-0.4.13.path=/home/mmphego/.arduino15/packages/esp8266/tools/esptool/0.4.13 -prefs=runtime.tools.mkspiffs.path=/home/mmphego/.arduino15/packages/esp8266/tools/mkspiffs/0.2.0 -prefs=runtime.tools.mkspiffs-0.2.0.path=/home/mmphego/.arduino15/packages/esp8266/tools/mkspiffs/0.2.0 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=/home/mmphego/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2 -prefs=runtime.tools.xtensa-lx106-elf-gcc-1.20.0-26-gb404fb9-2.path=/home/mmphego/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2 -verbose /tmp/untitled96212026.tmp/sketch_mar08a/sketch_mar08a.ino
Using board 'd1' from platform in folder: /home/mmphego/.arduino15/packages/esp8266/hardware/esp8266/2.4.2
Using core 'esp8266' from platform in folder: /home/mmphego/.arduino15/packages/esp8266/hardware/esp8266/2.4.2
WARNING: Category 'Device Control Sensors Sensor Measurement Liquid Water Flow Meter' in library FlowMeter is not valid. Setting to 'Uncategorized'
Detecting libraries used...
/home/mmphego/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -I/home/mmphego/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/tools/sdk/include -I/home/mmphego/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/tools/sdk/lwip2/include -I/home/mmphego/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/tools/sdk/libc/xtensa-lx106-elf/include -I/tmp/arduino_build_157338/core -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DARDUINO=10808 -DARDUINO_ESP8266_WEMOS_D1R1 -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_WEMOS_D1R1\"" -DESP8266 -I/home/mmphego/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266 -I/home/mmphego/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/variants/d1 /tmp/arduino_build_157338/sketch/sketch_mar08a.ino.cpp -o /dev/null
Generating function prototypes...
/home/mmphego/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -I/home/mmphego/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/tools/sdk/include -I/home/mmphego/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/tools/sdk/lwip2/include -I/home/mmphego/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/tools/sdk/libc/xtensa-lx106-elf/include -I/tmp/arduino_build_157338/core -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DARDUINO=10808 -DARDUINO_ESP8266_WEMOS_D1R1 -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_WEMOS_D1R1\"" -DESP8266 -I/home/mmphego/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266 -I/home/mmphego/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/variants/d1 /tmp/arduino_build_157338/sketch/sketch_mar08a.ino.cpp -o /tmp/arduino_build_157338/preproc/ctags_target_for_gcc_minus_e.cpp
/home/mmphego/Apts/arduino-1.8.8/tools-builder/ctags/5.8-arduino11/ctags -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives /tmp/arduino_build_157338/preproc/ctags_target_for_gcc_minus_e.cpp
Compiling sketch...
/home/mmphego/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++ -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -I/home/mmphego/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/tools/sdk/include -I/home/mmphego/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/tools/sdk/lwip2/include -I/home/mmphego/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/tools/sdk/libc/xtensa-lx106-elf/include -I/tmp/arduino_build_157338/core -c -Wall -Wextra -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11 -MMD -ffunction-sections -fdata-sections -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DARDUINO=10808 -DARDUINO_ESP8266_WEMOS_D1R1 -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_WEMOS_D1R1\"" -DESP8266 -I/home/mmphego/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266 -I/home/mmphego/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/variants/d1 /tmp/arduino_build_157338/sketch/sketch_mar08a.ino.cpp -o /tmp/arduino_build_157338/sketch/sketch_mar08a.ino.cpp.o
Compiling libraries...
Compiling core...
/home/mmphego/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-gcc -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -I/home/mmphego/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/tools/sdk/include -I/home/mmphego/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/tools/sdk/lwip2/include -I/home/mmphego/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/tools/sdk/libc/xtensa-lx106-elf/include -I/tmp/arduino_build_157338/core -c -g -x assembler-with-cpp -MMD -mlongcalls -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DARDUINO=10808 -DARDUINO_ESP8266_WEMOS_D1R1 -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_WEMOS_D1R1\"" -DESP8266 -I/home/mmphego/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266 -I/home/mmphego/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/variants/d1 /home/mmphego/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/cores/esp8266/cont.S -o /tmp/arduino_build_157338/core/cont.S.o
/home/mmphego/.arduino15/packages/esp8266/tools/xtensa-lx106-elf-gcc/1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-gcc -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ -I/home/mmphego/.arduino15/packages/esp8266/hardware/esp8266/2.4.2/tools/sdk/include -I/home/mmphego/.arduino15/packages/esp8266/hardware/esp826 [... cut ...]
└─ [2019-03-08 09:07:35] $ >>> ls ../7/liblto_plugin.so
../7/liblto_plugin.so
```
$ uname -a
```
Linux 4n0m41y 4.15.0-46-generic #49-Ubuntu SMP Wed Feb 6 09:33:07 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux

$ java --version
```
openjdk 10.0.2 2018-07-17
OpenJDK Runtime Environment (build 10.0.2+13-Ubuntu-1ubuntu0.18.04.4)
OpenJDK 64-Bit Server VM (build 10.0.2+13-Ubuntu-1ubuntu0.18.04.4, mixed mode)
```
$ lsb_release -a
```
Distributor ID: Ubuntu
Description:    Ubuntu 18.04.2 LTS
Release:    18.04
Codename:   bionic
```

facchinm, 2019-03-08:
Hi @mmphego ,
the issue is related with esp8266 toolchain, the right place to report this bug is https://github.com/esp8266/Arduino . Thanks!

closed 2019-03-08 by facchinm

mmphego, 2019-03-08:
Hi @facchinm , thanks for the feedback. I fixed the issue by deleting `~/.platfomio` and `~/.arduino`, then reinstalled platformio and arduino