arduino/Arduino #8624: [Library Manager] Request to add to Library Manager.

gmarty2000, 2019-03-08:
Good evening, could you add this library?
[https://github.com/gmarty2000/arduino-ULTRASONIC](https://github.com/gmarty2000/arduino-ULTRASONIC)
Thanks

per1234, 2019-03-08:
Hi. Each library added to the Library Manager index must have a unique `name` value in library.properties. There is already a library in the index with the `name` of "Ultrasonic". Please change the `name` value in your library.properties file to a unique value and then either create a new tag/release or else redo the existing tag/release.

To see which names are already taken, you can either look in the Arduino IDE's Library Manager (Sketch > Include Library > Manage Libraries) or the JSON file:
http://downloads.arduino.cc/libraries/library_index.json

After you have resolved this issue, please comment here so we'll know we can proceed with adding your library to the index.

More information:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ#how-can-i-add-my-library-to-library-manager

gmarty2000, 2019-03-08:
Fix it! I'm sorry for the inconvenience

cmaglie, 2019-03-11:
Done

closed 2019-03-11 by cmaglie