arduino/Arduino #8626: IDE hangs on compile Mac OSX Mojave

timwilson235, 2019-03-08:
MacOS Mojave 10.14.3, Arduino IDE 1.8.8. 

Can't compile ("verify") anything, as compile hangs at point "detecting libraries used". Happens with all my programs, including the simplest possible:

---------------------
void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}

------------------------

Tried several target boards, including the most basic Arduino/Genuino Uno.
Here is the verbose output:

-------------------------
/Applications/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/twilson/Documents/Arduino/libraries -fqbn=arduino:avr:uno -ide-version=10808 -build-path /var/folders/9r/856cqfbn0lb443dvjglw98zr0000gn/T/arduino_build_430604 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avrdude-6.3.0-arduino14.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avr-gcc.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avr-gcc-5.4.0-atmel3.6.1-arduino2.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.arduinoOTA.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.arduinoOTA-1.2.1.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -verbose /var/folders/9r/856cqfbn0lb443dvjglw98zr0000gn/T/untitled328350557.tmp/sketch_mar08a/sketch_mar08a.ino
/Applications/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Applications/Arduino.app/Contents/Java/hardware -tools /Applications/Arduino.app/Contents/Java/tools-builder -tools /Applications/Arduino.app/Contents/Java/hardware/tools/avr -built-in-libraries /Applications/Arduino.app/Contents/Java/libraries -libraries /Users/twilson/Documents/Arduino/libraries -fqbn=arduino:avr:uno -ide-version=10808 -build-path /var/folders/9r/856cqfbn0lb443dvjglw98zr0000gn/T/arduino_build_430604 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avrdude-6.3.0-arduino14.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avr-gcc.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avr-gcc-5.4.0-atmel3.6.1-arduino2.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.arduinoOTA.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.arduinoOTA-1.2.1.path=/Applications/Arduino.app/Contents/Java/hardware/tools/avr -verbose /var/folders/9r/856cqfbn0lb443dvjglw98zr0000gn/T/untitled328350557.tmp/sketch_mar08a/sketch_mar08a.ino
Using board 'uno' from platform in folder: /Applications/Arduino.app/Contents/Java/hardware/arduino/avr
Using core 'arduino' from platform in folder: /Applications/Arduino.app/Contents/Java/hardware/arduino/avr
Detecting libraries used...
/Applications/Arduino.app/Contents/Java/hardware/tools/avr/bin/avr-g++ -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega328p -DF_CPU=16000000L -DARDUINO=10808 -DARDUINO_AVR_UNO -DARDUINO_ARCH_AVR -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/cores/arduino -I/Applications/Arduino.app/Contents/Java/hardware/arduino/avr/variants/standard /var/folders/9r/856cqfbn0lb443dvjglw98zr0000gn/T/arduino_build_430604/sketch/sketch_mar08a.ino.cpp -o /dev/null

facchinm, 2019-03-11:
@gvarisco any idea? Could it be due to some Mojave security policy?

gvarisco, 2019-03-11:
Ouch, nothing sec/policy-related that I'm familiar with. @timwilson235 hi there! Could you please open the Console application (from the Utilities folder inside your Applications folder) and check if there's anything reported after pressing 'Verify'? The utility should open to All Messages, showing the log entries for everything that's happened recently on your Mac.

timwilson235, 2019-03-11:
I spent most of Sunday tracking this down - and SUCCEEDED! (much to my surprise). The problem was this old LaunchAgent in /Library/LaunchAgents: com.sourceforge.macgpg2.gpg-agent.plist. In fact, it seems that if this LaunchAgent is loaded, then NO 32-bit apps will run - they just silently hang - and all the Arduino tools (avr-g++, ctags, etc.) are 32-bit (that's gonna have to get fixed soon!).

I don't know the exactly how gpg-agent messes up the system; I only know that if it's loaded, Arduino won't compile, and if it's not loaded (ie moved out of the LaunchAgents folder) then Arduino works fine. 

So I guess this issue can be closed...although, I hope you guys are planning on making Arduino tools 64-bit, because soon they won't run at all (I expect you know this).

facchinm, 2019-03-11:
We already have a bunch of open issues on 64 bit only binaries on osx and we are porting all the tools (which are a lot, btw) to the new target. Until your report, however, it was only a small annoyance (some popup window here and there, just the first time you lauched a 32 bit tool).
@gvarisco @sandeepmistry @cmaglie should we speedup the recompile effort?

gvarisco, 2019-03-15:
@facchinm we definitely need to start working on that, but we've got still quite some time. According to Apple:
>>>
The 64-bit transition for macOS and macOS apps is still underway, so final transition dates have not yet been established. But now is a good time to check with the software developer to see if 64-bit versions of your favorite titles are available.

fabianonline, 2019-03-19:
Maybe it's another bug then... But I'm hit by it as well.

MacOS 10.14.3, Arduino 1.8.9, AVR Boards 1.6.23.

Trying to compile the minimal "default sketch":

```c
void setup() {
  // put your setup code here, to run once:

}

void loop() {
  // put your main code here, to run repeatedly:

}
```

Following appears in the Arduino console window:
```
/private/var/folders/zl/y_5l2b4n3_v3_z9d6ls430n40000gn/T/AppTranslocation/F0F048B8-FBD0-426B-A507-43D0D03313BD/d/Arduino 2.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /private/var/folders/zl/y_5l2b4n3_v3_z9d6ls430n40000gn/T/AppTranslocation/F0F048B8-FBD0-426B-A507-43D0D03313BD/d/Arduino 2.app/Contents/Java/hardware -hardware /Users/fabian/Library/Arduino15/packages -tools /private/var/folders/zl/y_5l2b4n3_v3_z9d6ls430n40000gn/T/AppTranslocation/F0F048B8-FBD0-426B-A507-43D0D03313BD/d/Arduino 2.app/Contents/Java/tools-builder -tools /private/var/folders/zl/y_5l2b4n3_v3_z9d6ls430n40000gn/T/AppTranslocation/F0F048B8-FBD0-426B-A507-43D0D03313BD/d/Arduino 2.app/Contents/Java/hardware/tools/avr -tools /Users/fabian/Library/Arduino15/packages -built-in-libraries /private/var/folders/zl/y_5l2b4n3_v3_z9d6ls430n40000gn/T/AppTranslocation/F0F048B8-FBD0-426B-A507-43D0D03313BD/d/Arduino 2.app/Contents/Java/libraries -libraries /Users/fabian/Documents/Arduino/libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10809 -build-path /var/folders/zl/y_5l2b4n3_v3_z9d6ls430n40000gn/T/arduino_build_174582 -warnings=none -build-cache /var/folders/zl/y_5l2b4n3_v3_z9d6ls430n40000gn/T/arduino_cache_952322 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=/private/var/folders/zl/y_5l2b4n3_v3_z9d6ls430n40000gn/T/AppTranslocation/F0F048B8-FBD0-426B-A507-43D0D03313BD/d/Arduino 2.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avrdude-6.3.0-arduino14.path=/private/var/folders/zl/y_5l2b4n3_v3_z9d6ls430n40000gn/T/AppTranslocation/F0F048B8-FBD0-426B-A507-43D0D03313BD/d/Arduino 2.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.arduinoOTA.path=/private/var/folders/zl/y_5l2b4n3_v3_z9d6ls430n40000gn/T/AppTranslocation/F0F048B8-FBD0-426B-A507-43D0D03313BD/d/Arduino 2.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.arduinoOTA-1.2.1.path=/private/var/folders/zl/y_5l2b4n3_v3_z9d6ls430n40000gn/T/AppTranslocation/F0F048B8-FBD0-426B-A507-43D0D03313BD/d/Arduino 2.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avr-gcc.path=/private/var/folders/zl/y_5l2b4n3_v3_z9d6ls430n40000gn/T/AppTranslocation/F0F048B8-FBD0-426B-A507-43D0D03313BD/d/Arduino 2.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avr-gcc-5.4.0-atmel3.6.1-arduino2.path=/private/var/folders/zl/y_5l2b4n3_v3_z9d6ls430n40000gn/T/AppTranslocation/F0F048B8-FBD0-426B-A507-43D0D03313BD/d/Arduino 2.app/Contents/Java/hardware/tools/avr -verbose /var/folders/zl/y_5l2b4n3_v3_z9d6ls430n40000gn/T/untitled92856715.tmp/sketch_mar19a/sketch_mar19a.ino
/private/var/folders/zl/y_5l2b4n3_v3_z9d6ls430n40000gn/T/AppTranslocation/F0F048B8-FBD0-426B-A507-43D0D03313BD/d/Arduino 2.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /private/var/folders/zl/y_5l2b4n3_v3_z9d6ls430n40000gn/T/AppTranslocation/F0F048B8-FBD0-426B-A507-43D0D03313BD/d/Arduino 2.app/Contents/Java/hardware -hardware /Users/fabian/Library/Arduino15/packages -tools /private/var/folders/zl/y_5l2b4n3_v3_z9d6ls430n40000gn/T/AppTranslocation/F0F048B8-FBD0-426B-A507-43D0D03313BD/d/Arduino 2.app/Contents/Java/tools-builder -tools /private/var/folders/zl/y_5l2b4n3_v3_z9d6ls430n40000gn/T/AppTranslocation/F0F048B8-FBD0-426B-A507-43D0D03313BD/d/Arduino 2.app/Contents/Java/hardware/tools/avr -tools /Users/fabian/Library/Arduino15/packages -built-in-libraries /private/var/folders/zl/y_5l2b4n3_v3_z9d6ls430n40000gn/T/AppTranslocation/F0F048B8-FBD0-426B-A507-43D0D03313BD/d/Arduino 2.app/Contents/Java/libraries -libraries /Users/fabian/Documents/Arduino/libraries -fqbn=arduino:avr:mega:cpu=atmega2560 -ide-version=10809 -build-path /var/folders/zl/y_5l2b4n3_v3_z9d6ls430n40000gn/T/arduino_build_174582 -warnings=none -build-cache /var/folders/zl/y_5l2b4n3_v3_z9d6ls430n40000gn/T/arduino_cache_952322 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.avrdude.path=/private/var/folders/zl/y_5l2b4n3_v3_z9d6ls430n40000gn/T/AppTranslocation/F0F048B8-FBD0-426B-A507-43D0D03313BD/d/Arduino 2.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avrdude-6.3.0-arduino14.path=/private/var/folders/zl/y_5l2b4n3_v3_z9d6ls430n40000gn/T/AppTranslocation/F0F048B8-FBD0-426B-A507-43D0D03313BD/d/Arduino 2.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.arduinoOTA.path=/private/var/folders/zl/y_5l2b4n3_v3_z9d6ls430n40000gn/T/AppTranslocation/F0F048B8-FBD0-426B-A507-43D0D03313BD/d/Arduino 2.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.arduinoOTA-1.2.1.path=/private/var/folders/zl/y_5l2b4n3_v3_z9d6ls430n40000gn/T/AppTranslocation/F0F048B8-FBD0-426B-A507-43D0D03313BD/d/Arduino 2.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avr-gcc.path=/private/var/folders/zl/y_5l2b4n3_v3_z9d6ls430n40000gn/T/AppTranslocation/F0F048B8-FBD0-426B-A507-43D0D03313BD/d/Arduino 2.app/Contents/Java/hardware/tools/avr -prefs=runtime.tools.avr-gcc-5.4.0-atmel3.6.1-arduino2.path=/private/var/folders/zl/y_5l2b4n3_v3_z9d6ls430n40000gn/T/AppTranslocation/F0F048B8-FBD0-426B-A507-43D0D03313BD/d/Arduino 2.app/Contents/Java/hardware/tools/avr -verbose /var/folders/zl/y_5l2b4n3_v3_z9d6ls430n40000gn/T/untitl [... cut ...]
standard	05:32:51.667912 +0100	taskgated	MacOS error: -67062
standard	05:32:51.932064 +0100	taskgated	MacOS error: -67062
standard	05:32:51.935196 +0100	syspolicyd	MacOS error: -67062
standard	05:32:51.944833 +0100	syspolicyd	MacOS error: -67062
fehler	05:32:51.948345 +0100	syspolicyd	URL failed validity check (-67062): <private>
standard	05:32:51.966258 +0100	syspolicyd	-[LSCodeEvaluationClientManager handlePromptResponse:info:identifier:]: response: 2 info: <private>
fehler	05:32:51.966482 +0100	syspolicyd	failed to call driver: 0x10000003
```

timwilson235, 2019-03-19:
Did you look to see if your Mac is running the gpg-agent LaunchAgent that was causing the problem for me?

fabianonline, 2019-03-20:
I have no gpg-agent running neither did I ever install one.

I've unloaded all LaunchAgents in ~/Library/LaunchAgents and /Library/LaunchAgents. Now the compiling reaches the "Sketch is being compiled..." phase before hanging...

timwilson235, 2019-03-20:
Try compiling after booting into Safe mode, see what happens.

fabianonline, 2019-03-20:
Booting in safe mode led to successful compiling. Booting normally after that also led to successful compiling. Interesting.

I did reboot normally in the meantime; that didn't work. Apparently there is something fixing itself by trying to compile in safe mode or something...

timwilson235, 2019-03-21:
Yeah, I saw some very strange behavior also when debugging this. But since compile works in safe mode, then it's probably down to SOME LaunchAgent or LaunchDaemon. Is there anything in /Library/LaunchAgents, /Library/LaunchDaemons, or ~/Library/LaunchAgents that looks particularly old? The one that was causing my problem was dated 2009 - that should've been a red flag right away.

facchinm, 2019-03-21:
@gvarisco @sandeepmistry as OSX gurus, do you have any idea of what is possibly going on :slightly_smiling_face: ?

Coder-256, 2019-04-23:
Have you made sure that you dragged the app to disk? I am asking because of App Translocation, which can cause a ton of problems, and is in fact mentioned in @fabianonline's log. [Here is a good explanation](https://github.com/potionfactory/LetsMove/issues/56#issue-162975492). However I don't have even the slightest clue how GPG could interfere with any of this, so I could be totally wrong. Plus, it seems that @timwilson235 had already moved the app out of translocation. Anyway it is worth seeing that, if you can reproduce the issue, whether dragging the app to `/Applications` before launching it helps anything.

timwilson235, 2019-04-23:
Yes, the app was in /Applications.

newkton1, 2019-05-19:
With 10.14.6 beta 1, I'm getting occasional serial port not found error messages from 1.8.9. 
macos 10.15 will be shown at WWDC in 3 weeks and requires 64-bit applications—the GM release will be September/October if it follows the usual release shedule for beta testing.  That leaves about 16 weeks to port the 32-bit Arduino 1.8.9 to 64-bit or there'll be problems for mac users. Is somebody working on this?

matttarantino, 2019-06-07:
MacOS 10.15 beta user here. I am now getting `bad CPU type in executable` which makes sense. @gvarisco I would be happy to beta test once you are closer to release!

jamesarm97, 2019-06-08:
Same here. I need to be able to run the 10.15 beta for development but also need aurduino compiling capability. I would definitely love to help beta test 64 but OS X Arduino tools.

iammosespaulr, 2019-06-09:
Mee Tooo!!

garymansted, 2019-06-11:
> Hey @newkton1, we're aware of it and are tracking this internally. Once I have an ETA to share I'll get back to you.

I got trigger happy and went the for 10.15 beta as well - `bad CPU type in executable`. Plz keep me in the loop too.

facchinm, 2019-06-14:
:wave: 
This PR (https://github.com/arduino/Arduino/pull/8976) bundles the new avr-gcc built for 64 bit macOS; it would be awersome if you could try it (on 10.15 but also on older/other OS) and report any abnormal behaviour you may find.
`ctags` has not been rebuilt yet but will be soon

sgreensides, 2019-08-24:
Just bought MKR 1010 Wifi and the MacOS Catalina / Arduino IDE Beta (64bit) does not find it.  Board not found

sgreensides, 2019-08-24:
File / device "/dev/ttys000" changes to the same date & time as when the Arduino MKR 1010 Wifi is plugged in to the USB port

facchinm, 2019-08-26:
@sgreensides the board not being recognized is another kind of issue. Could you try changing USB port / cable ? On OSX the board should appear as something like `/dev/tty.usbmodem1412`

cpatulea, 2020-09-20:
Mac OS 10.15.6, Arduino 1.8.13, ESP32 based board - seems to work.

per1234, 2021-03-28:
Closing as resolved by the completion of the transition to 64 bit binaries for macOS hosts.

closed 2021-03-28 by per1234