arduino/Arduino #8632: Can't change editor font in Arduino on linux mint

MaiMariarti, 2019-03-09:
I'm running Linux Mint 19.1 Tessa
I changed the "editor.font" inside preferences.txt but can't get any of the fonts that are installed on this system to work.
My windows 7 installation works fine.

per1234, 2019-03-11:
Hi. I just tested on Linux and changing fonts in preferences.txt is working fine for me.

Please make sure you are using the preferences.txt at the path shown on the line following **File > Preferences > More preferences can be edited directly in the file**.

Note that you must all Arduino IDE windows before editing preferences.txt. Changes you make to preferences.txt will not take effect and will also be lost when you close the Arduino IDE.

Let me know if that resolves your problem. If not, please state which version of the Arduino IDE you're using.

MaiMariarti, 2019-03-11:
Hi. Thanks for getting back to me.
Forgive me for saying so, but having 2 references.txt and having one of them be there for no reason, is just dumb programming.
There is a useless version in the installation folder "/opt/arduino-1.8.8/lib" and one at "/home/'username'/.arduino15"!!
Who thought this would be a good idea? I wasted over an hour on this stupid problem and moved over to Atom. Don't want this kind of stupidity to happen again. 
This is what happens when people are allowed to do whatever the hell they feel like. 
Its bad enough that the IDE is so basic for such a popular development environment, having preferences be edited in a text file where it could've easily been implemented in the preferences section of the gui, like all other IDEs on the market, shows how unorganized and retarded the project really is.  
To top it off, the program writing over the file at the time of exit, is doubly stupid.
I am sure some one would say "Why don't you help fix it if you don't like it, after all this is an open project, instead of ranting about it" ! There are always stupid illogical arguments like that when confronted by legitimate complains.
God knows how many people were screwed over this stupid problem. I got the Atom environment working and hopefully I'll never have to come back to this retarded program ever again.  
I know it may not be your fault, but want my protest to be public record.

closed 2019-03-11 by MaiMariarti