arduino/Arduino #8636: [Library Manager] Please *RE* add BNO055 server for SimplePacketComs library (i fixed the conflicts)

madhephaestus, 2019-03-11:
https://github.com/madhephaestus/BNO055SimplePacketComs

This library adds Simple Packet Coms server of the BNO055 IMU module. The server is not platform dependent and will work with any implementation of SimplePacketComs

This library was providing Adafruit_Sensor.h and was causing compilation conflicts. It appears it was de-listed as a result. I have fixed the dependency issue and made a repaired release.

madhephaestus, 2019-03-11:
The upstream Adafruit Universal Sensor library merged and released the fix on their side, and my side is compatible with it. 

https://github.com/adafruit/Adafruit_Sensor/pull/26

cmaglie, 2019-03-18:
Done!

madhephaestus, 2019-03-18:
@cmaglie Thank you! I do have one question, is there a way to get notified when a library one has published gets de-listed? I would have fixed this earlier if i had known it was causing issues.

cmaglie, 2019-03-28:
There isn't an official way to get notified, cases like yours are very rare (I remember only another one) so we didn't think about a procedure.

We should have contacted you before delisting the library (via github or the email written in library.properties...) sorry about that!

madhephaestus, 2019-04-01:
Well if its so rare i guess having one-off procedures is ok. Thanks for reinstating this :)

closed 2019-04-01 by madhephaestus