arduino/Arduino #8649: please add a new Class for Arduino

offcircuit, 2019-03-13:
https://github.com/offcircuit/Morse

Thanks,
Helder Rodrigues

per1234, 2019-03-15:
In order to be added to the Library Manager index your library must contain a library.properties file. The format of this file is documented here:
https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification#libraryproperties-file-format

Make sure that the `version` value matches the tag of the release that contains it.

Once you have done this please comment here so we know we can proceed.

For more information see:
https://github.com/arduino/Arduino/wiki/Library-Manager-FAQ

offcircuit, 2019-03-15:
Hello again,

it is done.

thanks,
Helder Rodrigues

per1234, 2019-03-15:
Hi. Unfortunately, the `version` value in your library.properties is not compliant with the library specification:
>Version should be [semver](http://semver.org/) compliant. 1.2.0 is correct; 1.2 is accepted; r5, 003, 1.1c are invalid

Please change the `version` value to a valid version number and then either create a new tag/release or else redo the existing tag/release.

Once you have done this, please comment here so we'll know we can proceed.

offcircuit, 2019-03-15:
... and complete ...

Thanks,
Helder Rodrigues

offcircuit, 2019-03-15:
Hi again can I add a new class:

https://github.com/offcircuit/SHCSR04

Thanks,
Helder Rodrigue

offcircuit, 2019-03-17:
Dear Sir, 

I'm having a problem with the switch / case condition at the end of the statement. I have to get around, to run well.
Returns a memory problem.

But before ordering the values does not present this error.

Please see the files at:
https://github.com/offcircuit/char57

Can you help me with this dilemma?

Thanks,
Helder Rodrigues

per1234, 2019-03-17:
>I'm having a problem with the switch / case condition at the end of the statement.

If you haven't done so already, please post about the issue to the [Arduino Forum](http://forum.arduino.cc/). That is a more appropriate place for this topic. I'm sure we'll be able to help you over there.

cmaglie, 2019-03-18:
Done

closed 2019-03-18 by cmaglie