arduino/Arduino #8654: 01. Basics -> Formatted Minimum

JacobChrist, 2019-03-14:
Dear Arduino Community,

I teach a community college class using the Arduino IDE with chipKIT boards.  One thing that would be helpful to the students is to have a few more comments in a blank project when they start up.  I think it would be great if this replaced the Bare Minimum example, but a next best would be to add a new example sketch to the 01. Basics sketches potentially called "Formatted Minimum" or some better name that I'm sure someone will come up with.

Thanks for your consideration,

Jacob

``` c++
// put #include and other compiler preprocessors here

// put variables of scope global here

void setup() {
  // put variables of scope local here:

  // put your setup code here, to run once:

}

void loop() {
  // put variables of scope local here:

  // put your main code here, to run repeatedly:

}

// put user defined functions here: Do not put functions inside functions

void function_1() {
  // put variables of scope local here:

  // put your function_1 code here:

}

void function_2() {
  // put variables of scope local here:

  // put your function_2 code here:

}

Pieter12345, 2019-03-20:
@JacobChrist The BareMinimum sketch is what opens when you create a new sketch. Anything that will be added to this sketch is forced upon every user (beginner to expert). Personally, I would dislike to have to remove a few functions every time I create a new sketch.

Regarding adding a new Basic sketch for explaining how functions work: This could be done, but this seems to focus on how C works, whereas all other sketches seem to focus on Arduino specific functionality. It is up to the developers of this project to determine where to put this border.

Considering that you are teaching students, I would recommend supplying these students with a sketch that contains what you think the students need for the course that you are giving. This allows you to go as far as providing a code skeleton in which students will only have to (partially) implement already-existing functions.

dsyleixa, 2019-03-21:
I double  Pieter12345's  comment, I would not like to have that FormattedMinimum.ino specimen as a general substitute to  bareminimum.ino . 
As a workaround, you might wish change and save bareminimum.ino to suit to your own personal requirements.
As to mefunction_1 and function_2  some comments:

1st, as both are supposed to be called from setup() or loop(), they should be located at top of setup() or loop() to avoid requirung prototypes (it's a bad attititude to force the IDE to do that)
2nd, most of the needed extra functions commonly are NOT of type "void" and also require  additional  arguments to be passed to, so I actually doubt that the FormattedMinimum.ino specimen would be so much helpful at all.

Just my 2 ct 8)

JacobChrist, 2019-03-27:
@Pieter12345 Fair enough, but it is my goal to force my will on every user.

@dsyleixa Don't get me started on bad habits of the Arduino IDE.

closed 2019-03-27 by JacobChrist