arduino/Arduino #8656: Bundled jssc-arduino

kevans91, 2019-03-15:
Hi,

Is the source for the bundled jssc-arduino made available anywhere? We have some [FreeBSD patches for JSSC](https://svnweb.freebsd.org/ports/head/comms/java-simple-serial-connector/files/) (up-to-date version) that [have failed to go upstream](https://github.com/scream3r/java-simple-serial-connector/pull/81) (slightly older version) -- it would be nice to get these applied to the version you bundle since they're a rather minimal set we generally use the same interface as Linux.

I also have an accompanying [Arduino patch](https://svnweb.freebsd.org/ports/head/devel/arduino18/files/patch-arduino-core_src_processing_app_SerialPortList.java?view=markup) that we had to apply to the SerialPortList here in this repository, that matches the equivalent file on JSSC.

facchinm, 2019-03-27:
Hi Kyle,
the binary we provide is simply a compiled version of https://github.com/scream3r/java-simple-serial-connector . The suffix `-arduinoX` is due to the different compile flags we have been using for the native JNI part. 
Since that project is basically dead and we are gradually switching to a collection of modular (golang based) tools it's probably a better idea to wait a bit and concentrate the efforts on making the new tool FreeBSD compatible from the beginning.

sandeepmistry, 2019-09-16:
I'm closing this as it seems @facchinm has answered the question.

Please re-open if any additional discussion is needed.

closed 2019-09-16 by sandeepmistry