arduino/Arduino #8657: strange compile behaviour with function name containing "setup"

johnty, 2019-03-15:
I'm facing a bizzare compile error that happens under very specific situations. It relates to using the OSC library, but I'm not sure if its something specific to the library or something funny going on with the Arduino build process. The code below with comments contain the necessary steps to reproduce the compile error.

Basically, under _certain_ conditions if we have a function called **setup**Foo or Foo**setup** or whatever in a second .ino file tab, the compile fails. If you declare it in the main sketch file everything works fine. Doing certain seemingly unrelated things (like having another function called **setup**Bar in the main sketch), causes it to compile without issues. My guess is it relates to something the build process is handling any function names with "setup" in it, but it could just be a red herring...

Build targets are esp8266 and esp32, configured using board manager (although I think setting them up via the git repo doesn't make any difference either).

Any comments, or pointers in the right direction would be appreciated!

```
#include <WiFiUdp.h> //test #1-4
#include <OSCMessage.h> // https://github.com/CNMAT/OSC

//test #6 - uncomment setupBarrr()
//void setupBarrr() {  
//}

void setup() {
  delay(10);
}

void loop() {
  delay(10);
}

void fromHost(OSCMessage &msg, int addressOffset) {
}

//test #1-4 -> move this to other tab
void Foosetup() { //could be setupFoo, setupBlah, etc... same behaviour
   delay(10);  
}

// Test Cases:
//1. #include <WiFiUdp.h> + Foosetup in another_file.ino == fail
//2. comment out line 1: #include <WiFiUdp.h> + Foosetup in another_file.ino == fail
//3. remove line 1: #include <WiFiUdp.h> + Foosetup in another_file.ino == OK
//4. blank 1st line + Foosetup in another_file.ino == fail
//5. Foosetup in this file + any of the above == OK
//6. setupBarrr() in this file + any of the above == OK

// error message for fail cases: 
// weird_arduino_bug:16:15: error: variable or field 'fromHost' declared void
// void fromHost(OSCMessage &msg, int addressOffset) {
//               ^
// weird_arduino_bug:16:15: error: 'OSCMessage' was not declared in this scope
// weird_arduino_bug:16:27: error: 'msg' was not declared in this scope
// void fromHost(OSCMessage &msg, int addressOffset) {
//                           ^
// weird_arduino_bug:16:32: error: expected primary-expression before 'int'
// void fromHost(OSCMessage &msg, int addressOffset) {
```

facchinm, 2019-03-15:
Hi @johnty ,
all the bug you are reporting should be fixed in the latest IDE. 
Make sure you are running IDE 1.8.9 or today's hourly and please reopen the issue if it still fails, thanks

closed 2019-03-15 by facchinm

johnty, 2019-03-15:
Thank you for the quick response! 

I tried the hourly and it seems to be fine now. Could you kindly point me to exactly where the fix was, or even a very basic description of what was happening that caused this? I took many hours to reproduce this behaviour (had to whittle down a large project piece by piece to get to this point), and it would be nice to learn about how things are working under the hood.

facchinm, 2019-03-15:
It was probably this commit (https://github.com/arduino/arduino-builder/commit/491a06767ebe5d0b03b9ddf0485de565e00a976f) which has been included in arduino-builder 1.4.2 , merged here https://github.com/arduino/Arduino/commit/4b6b8f0c52402a6b55efdcd3cd26ea8c1c14bd59

johnty, 2019-03-15:
thank you very much!