arduino/Arduino #8659: Setting $HOME/Documents/Arduino as systemwide default sketchbook folder

ysalmon, 2019-03-15:
I am currently trying to deploy the Arduino IDE to a bunch of Linux machines in a teaching environment. (disclaimer : I am not an Arduino user myself)

I am facing some difficulty with the way the IDE selects the sketchbook folder for a new user. In my tests it is located in `~/Arduino`, and this is consistent with my understanding of [the corresponding method in the Platform class for Linux](https://github.com/arduino/Arduino/blob/afc0e0240e966b3bb7a9527d60566adbf76ee7ac/arduino-core/src/processing/app/linux/Platform.java#L50).

I suggest it would be interesting to use `$(xdg-user-dir DOCUMENTS)/Arduino` which by default would be `~/Documents/Arduino`. I am not proposing xdg-compliance just for the sake of it : in our concrete working environment with both Windows and Linux workstations, `~/Documents` (and what is under it) is mounted to a CIFS network share that is accessible from Windows workstations as well, while `~` and `~/Arduino` cannot be.

I tried to tweak the `sketchbook.path` preference in the global default `preferences.txt` but it does not seem to support variables like `~/Documents/Arduino` or `$HOME/Documents/Arduino`.

I understand there has already been discussions about xdg directory structure compliance (#3915) and the migration problems a change would cause (https://github.com/arduino/Arduino/issues/7651#issuecomment-393703658).

So my suggestion is to maybe not make this default behaviour but allow for a `xdg-compliance` preference in the default, global `preferences.txt` (would be ignored if present in a user's file) that a system administrator could enable. This would make it possible to leave existing installations untouched while progressively adapting things, in a manner consistent with e.g. [the linux install script](https://github.com/arduino/Arduino/blob/master/build/linux/dist/install.sh) which goes at lengths to use the xdg API.

Different but related : what is the rationale for [reverting to the default sketchbook folder](https://github.com/arduino/Arduino/blob/0e45f4e0d4d1ff9a2d11be8140834c7ad8d428e6/arduino-core/src/processing/app/BaseNoGui.java#L361) when the folder set in the preferences does not exist, instead of creating it (as is done for the default path anyway) ?

lhanson, 2019-10-17:
Please comply with users' XDG preferences, there is no reason to be creating yet another junk-drawer directory in a user's `$HOME` and numerous reasons not to.

lhanson, 2019-10-17:
Other conversation on the topic here: https://github.com/arduino/Arduino/issues/3915

ysalmon, 2019-10-17:
BTW if somebody is facing the same problem as me, here is an ugly workaround pending resolution of this :

after installing arduino IDE on the machine : ``mv /opt/arduino/arduino /opt/arduino/arduino.real`` then create an executable shell script ``/opt/arduino/arduino`` with content :
```bash
#!/bin/bash
mkdir -p $HOME/.arduino15
if [ ! -f $HOME/.arduino15/preferences.txt ]; then
  mkdir -p $(xdg-user-dir DOCUMENTS)/Arduino
  echo "sketchbook.path=$(xdg-user-dir DOCUMENTS)/Arduino" > $HOME/.arduino15/preferences.txt
fi
/opt/arduino/arduino.real $@
```

On first run by the user, this will create an appropriate default in the preferences.txt file.

rhysday, 2021-09-15:
Arduino creates THREE folders in my HOME directory. Even if I create a wrapper and try to change HOME location it doesn't work because it's a java app.