arduino/Arduino #8661: internal compiler error: in lto_output_varpool_node

MrSVCD, 2019-03-16:
I am trying to compile https://github.com/dhansel/Altair8800 from 10 of march and I get this error with arduino avr core 1.6.22:

lto1: internal compiler error: in lto_output_varpool_node, at lto-cgraph.c:624
Please submit a full bug report,
with preprocessed source if appropriate.
See <http://gcc.gnu.org/bugs.html> for instructions.
lto-wrapper: fatal error: /home/ghostwolf/Program/arduino-1.8.9/hardware/tools/avr/bin/avr-gcc returned 1 exit status
compilation terminated.
/home/ghostwolf/Program/arduino-1.8.9/hardware/tools/avr/bin/../lib/gcc/avr/5.4.0/../../../../avr/bin/ld: error: lto-wrapper failed
collect2: error: ld returned 1 exit status
exit status 1
Error compiling for board Arduino/Genuino Mega or Mega 2560.

MrSVCD, 2019-03-16:
Arduino AVR Boards version 1.6.21 does compile the code without errors.

MrSVCD, 2019-03-16:
I am running x86-64 Ubuntu-mate 18.04. Sorry for not including that from the start.

per1234, 2019-03-16:
Closing as duplicate of https://github.com/arduino/ArduinoCore-avr/issues/39

Also related: https://github.com/arduino/Arduino/issues/7949

The issue is specific to the avr-gcc 5.4.0 used by Arduino in Arduino AVR Boards 1.6.22 and 1.6.23. As you discovered, a reasonable workaround is to use Arduino AVR Boards 1.6.21 until a new release of Arduino AVR Boards comes out that doesn't use the problematic avr-gcc version.

There is also a workaround for your particular error described here:
https://github.com/arduino/ArduinoCore-avr/issues/39#issuecomment-418202150

closed 2019-03-16 by per1234